brocaar/chirpstack-network-server #9: Please help testing US 902-928 ISM band

brocaar, 2016-04-22:
I'm looking for people who are up for testing the US 902-928 build of LoRa Server. All the work has been done, except that I'm unable to verify if it is really working 😉  (I don't have hardware to test with for this band).

Band specific configuration can be found in: https://github.com/brocaar/lorawan/tree/master/band

Same installation instructies apply (https://github.com/brocaar/loraserver/wiki/Getting-started), except that you need a ``loraserver`` binary for the US band. For testing, you can find pre-compiled binaries for the US band here: https://www.dropbox.com/sh/kar32nvpeynm33h/AAB1bz1qu9YYUMw6zqbgiDDua?dl=0

To build US band binaries yourself (from ``master``):
```bash
BANDS=us_902_928 GOOS=linux GOARCH=amd64 make package
BANDS=us_902_928 GOOS=linux GOARCH=386 make package
BANDS=us_902_928 GOOS=linux GOARCH=arm make package
BANDS=us_902_928 GOOS=darwin GOARCH=amd64 make package
BANDS=us_902_928 GOOS=windows BINEXT=.exe GOARCH=386 make package
BANDS=us_902_928 GOOS=windows BINEXT=.exe GOARCH=amd64 make package
```

gzwsc2007, 2016-04-25:
Hi Orne,

I am Anson from Carnegie Mellon University, and I am working on a LoRaWAN-related research project. I am happy to test out the US915 band for you. Will let you know in a few days. BTW great work on the server! We have been looking for open-source LoRaWAN server for a long time!

Thanks,
Anson

Mehradzie, 2016-04-26:
I don't think US people would be able to receive packets in this stage. However, since Aussie frequencies overlap the US ones, I have put a US firmware on our gateway and end-nodes and tested the US frequencies with the changes that I did on lorawan and the test went alright. 

So probably will wait to hear from our US contributes to see if it works for them :D

brocaar, 2016-04-26:
Will re-compile binaries with the changes from @Mehradzie tonight. As @Mehradzie mentioned, the current binaries are broken.

brocaar, 2016-04-26:
@gzwsc2007 I've updated the binaries with @Mehradzie his fix (I forgot to multiply all durations by ``time.Second`` so all durations were in nanoseconds instead of seconds). I'm looking forward to your feedback!

https://www.dropbox.com/sh/troh10y9sap2h7b/AADZmtUHgGKse58XIpLwcSqpa?dl=0

gzwsc2007, 2016-04-26:
@brocaar I fixed a minor typo about DR0 of the US band, and have created a pull request under `lorawan`. With that fix, OTAA and confirmed data up/down link seems to be working.

I am using the stock (and dumb) Semtech packet forwarder, and the client node is a STM32 nucleo board + SX1276 shield, running the Semtech mbed demo app. (https://developer.mbed.org/teams/Semtech/code/LoRaWAN-demo-76/)

brocaar, 2016-04-27:
Cheers for testing and fixing the typo! I've merged your ``lorawan`` pull-request.

closed 2016-04-27 by brocaar