brocaar/chirpstack-network-server #10: Please help testing AU 915-928 band

brocaar, 2016-04-23:
Same as #9 except that this issue is for the AU 915-928 ism band 😄 

Pre-compiled binaries for testing can be found here:
https://www.dropbox.com/sh/ywg5v3469zct7i9/AABrkYi87ov-vIp1UC1e4CBWa?dl=0

Alternatively, you can build them from the ``au_915_928`` branch with the following commands:
```bash
BANDS=au_915_928 GOOS=linux GOARCH=amd64 make package
BANDS=au_915_928 GOOS=linux GOARCH=386 make package
BANDS=au_915_928 GOOS=linux GOARCH=arm make package
BANDS=au_915_928 GOOS=darwin GOARCH=amd64 make package
BANDS=au_915_928 GOOS=windows BINEXT=.exe GOARCH=386 make package
BANDS=au_915_928 GOOS=windows BINEXT=.exe GOARCH=amd64 make package
```

Mehradzie, 2016-04-26:
Under test.

Mehradzie, 2016-04-26:
Hi Orne,

I have done some testing with a Multitech Conduit and mDot end-nodes in Australian frequencies. I can't confirm the join process as our end-node might have some bugs in it which prevents the join process to be completed, however, if the credentials manually fed to the end-node, the RX and TX transmit is fully working (obviously after the changes I have submitted to Lorawan about the RX timestamp).

Here you go :)

Cheers,

Mehrad

brocaar, 2016-04-26:
Nice! Did you see any errors in the LoRa Server logs when trying an OTAA attempt? I had some difficulties with a iM880A-L node where it turned out some fields needed to be set LSBF, and some as MSBF (https://github.com/brocaar/loraserver/wiki/Activating-nodes#im880a-l--im880b-l).

Anyway, great to hear that the RX and TX is working after ABP!

closed 2016-05-08 by brocaar

Mehradzie, 2016-05-12:
Just coming back with some update.

I have tested the OTAA with our latest firmware on the end-nodes and I can say now that **it fully working**

I have test the OTA join on window 1 and 2 with both JOIN_ACCEPT_DELAYs and it was successful and robust. 
This bring the Australian branch functionality to the fully working and tested state :100: