brocaar/chirpstack-network-server #11: Could not decode RxPacket: invalid characters

chunyeow, 2016-04-24:
loraserver_1    | time="2016-04-24T14:47:08Z" level=error msg="gateway/mqttpubsub: could not decode RXPacket: invalid character 'ÿ' after top-level value" data_base64="Mv+HAwEBCFJYUGFja2V0Af+IAAECAQZSWEluZm8B/4oAAQpQSFlQYXlsb2FkAf+OAAAA/7H/iQMBAQZSWEluZm8B/4oAAQ0BA01BQwH/hAABBFRpbWUB/4YAAQlUaW1lc3RhbXABBgABCUZyZXF1ZW5jeQEIAAEHQ2hhbm5lbAEGAAEHUkZDaGFpbgEGAAEJQ1JDU3RhdHVzAQQAAQpNb2R1bGF0aW9uAQwAAQhDb2RlUmF0ZQEMAAEEUlNTSQEEAAEHTG9SYVNOUgEIAAEEU2l6ZQEGAAEIRGF0YVJhdGUB/4wAAAAR/4MGAQEFRVVJNjQB/4QAAAAQ/4UFAQEEVGltZQH/hgAAACf/iwMBAQhEYXRhUmF0ZQH/jAABAgEETG9SYQEMAAEDRlNLAQYAAAAW/40FAQEKUEhZUGF5bG9hZAH/jgAAABD/jwYBAQRNSERSAf+QAAAAE/+RBgEBB1BheWxvYWQB/5IAAAAO/5MBAQL/lAABBgEIAABm/4gBAQh/BAYISwIAAAL8sZP9rAH9JItAAQIBAQECAQRMT1JBAQM0LzUBeQH4ZmZmZmZmIEABHQEBCVNGMTJCVzEyNQAAAR4BgEEl6KEAAwACXo9uywBsXNfq4m6UHgUzdutLLeYA" 

What could be the problem for above errors?

ivajloip, 2016-04-24:
I think I saw such error when I forgot to update `lora-semtech-bridge` (while updating the `loraserver`). Are you running both latest `loraserver` and `lora-semtech-bridge` or what is your setup?

chunyeow, 2016-04-24:
I am using the docker-compose to run the latest loraserver. I saw the following:

semtechbridge:
  image: brocaar/lora-semtech-bridge
  links:
    - mosquitto
  environment:
    - MQTT_SERVER=tcp://mosquitto:1883
  ports:
    - "1700:1700/udp"

Is that going to use the latest image or any problem could be?

ivajloip, 2016-04-24:
I did not find *lora-semtech-bridge* in hub.docker.com, so could you give me the output of `docker images`? It seem to me that you might be using an old version of the docker image (maybe a new one is not available online). To build it you can obtain the code and run

    docker rm loraserver_semtechbridge_1
    docker build -t "brocaar/lora-semtech-bridge:latest" <folder where you have put the semtech bridge>
    docker up

chunyeow, 2016-04-25:
I remove the docker images dated back 8 weeks ago and rebuild the latest lora-semtech-bridge and it works. 

Thanks for your help.

closed 2016-04-25 by chunyeow