brocaar/chirpstack-network-server #13: [GOLANG] Importing local packages but with the "github.com/brocaar/" prefix?

gzwsc2007, 2016-04-26:
I am new to Go, so please pardon me if this question is stupid.

When I pull down the `loraserver` repo, and do `make`, I get errors like
`internal/loraserver/interfaces.go:4:2: cannot find package "github.com/brocaar/loraserver/models" in any of: ........`

`internal/loraserver/loraserver.go:12:2: cannot find package "github.com/brocaar/loraserver/models" in any of:.........`

If I change all the `"github.com/brocaar/loraserver/....."` to `"loraserver/......"`, then the code compiles.

I am using GO 1.6 and my GOPATH and GOROOT are setup correctly. What I don't understand is that, why are we prepending `loraserver` with `github.com/brocaar/`? Isn't `loraserver` a local package instead of a vendor package?

Thanks in advance.

brocaar, 2016-04-27:
This is because when you would do a ``go get github.com/brocaar/loraserver/...``, Go would place all the source-code under ``$GOPATH/src/github.com/brocaar/loraserver/...``. In Go it is common to use the full repo url as import path. In case of LoRa Server, all requirements are vendored, if that would not be the case a simple ``go get ./...`` would fetch all requirements for you 😄 

When you move your clone to the above path, everything should work.

closed 2016-04-27 by brocaar

gzwsc2007, 2016-04-27:
Thank you very much for answering my question! :D

sandeepmeenuga, 2019-07-31:
Is there anyway to change the import names from "github.com/xxx/xxx" to "common/xxx"(something as I want). My requirement is, I don't want to see these github.com/xxx/xxx in the build file to avoid the complete details about what I'm using if I open the go build file in text editor.

NguyenKyThinh94, 2021-08-29:
i think github and gitlab donate for go team a huge money :v