brocaar/chirpstack-network-server #15: Make FRMPayload decryption optional

ivajloip, 2016-05-09:
According to the spec (section 4.3.3.2), it should be possible to perform encryption on upper layers. In this case the network server should not try to decrypt the information. 

I am working on this right now. Currently I do not intend to support encryption preference for every possible port number as it seems required by the spec. For each device it will be possible to have the server either decrypt the payload or leave it as it is - the same behavior for all fport numbers other > 0 (obviously the default behavior will not change). 

How does this sound to you?

brocaar, 2016-05-09:
The section below has been dropped in the 1.0.1 version of the LoRaWAN specs:

> As a default, the encryption/decryption is done by the LoRaWAN layer for all FPort. The
encryption/decryption may be done above the LoRaWAN layer for specific FPorts except 0...

ivajloip, 2016-05-15:
Thanks for pointing me out that there is a new spec version (I have only found draft 3, but I guess that the final version will be quite similar)!

closed 2016-05-15 by ivajloip