brocaar/chirpstack-network-server #18: ERR unknown command 'PEXPIRE'

Daniel-dk, 2016-05-11:
HI, I'm getting an Error from what seems to be my gateway rx packet 
output from loraserver.
`INFO[0025] gateway/mqttpubsub: rx packet received`
`ERRO[0025] processing rx packet error: add rx packet to collect set error: ERR unknown command 'PEXPIRE'  data_base64=...`

the endnode also fails to join the network and reports a Join error

do we need to register the gateway with loraserver ?
Gateway is a Multitech Conduit
endnode is a Multitech mDot

brocaar, 2016-05-11:
It is a Redis error (http://redis.io/commands/pexpire). My guess is that you're running a Redis version < 2.6.0 as ``PEXPIRE`` is available since Redis 2.6.0. I'll update the docs with the minimum Redis version that must be used.

Please let me know if this fixes your issue.

Daniel-dk, 2016-05-11:
installing redis-2.8.19 worked.
Thank you :)

closed 2016-05-11 by Daniel-dk