brocaar/chirpstack-network-server #20: Feature: Web interface and API authentication and authorization

ivajloip, 2016-05-15:
Hi,
I think it will be helpful at some point of time to add some access control to the web API. As I am going to need that soonish, I have started looking into possible solutions. So far the use of [JWT][] seems as a good option - easy to integrate, does not require too much shared knowledge if someone wants to run multiple loraservers, etc. Will you be willing to have such functionality and do you have any idea how you would want it to be implemented?

[JWT]: https://en.wikipedia.org/wiki/JSON_Web_Token

peterbaugh, 2016-05-16:
I agree that access control needs implementing, I was thinking this very thing this morning and was going to post. You beat me to it :) Having implemented a token based system on another API it seams like the better option, easy to use and configure.

Also I would like to see the ability to run the server without the web front end, just configure everything through API calls. I do not know if this is possible.

brocaar, 2016-05-16:
@peterbaugh the web-frontent is just a simple AngularJS application on top of the API, so everything you can do in the web-interface, is possible through the API 😄 

One issue I'm having with adding access control is that the use-cases might be totally different per user. Maybe it is better to have a middleware layer on top of the LoRa Server API to expose certain API methods to your users. That way you have all the flexibility to customise it to your own needs (e.g. give certain users access to a limited set of nodes per application etc..).

Looking forward to your feedback!

ivajloip, 2016-05-17:
I understand that the use-cases might be totally different for each user and in that case everyone is free to create an interface (most likely web interface) and internally to query the API that you have provided. This is how I understand your intentions. My concern is that to my experience people are not feeling quite good if everyone on a given network can access some APIs. In this regard, a token based authentication for your APIs can

1. limit the access so that the security will be improved.
2. add almost no additional cost for your application as anyone with valid token can be given *admin* rights (full access).
2. add almost no additional cost for the API consumer as all that will be needed is the user of loraserver that has created an (web) interface, to add a HTTP header option with a valid token that expires in 10 years and forget about it (if he is really lazy).
3. people will not have to modify the loraserver if they are not concerned that much by the performance (as opposed to having to modify it in order to add some security layer).

My 2c :)

brocaar, 2016-05-17:
I'll have a look into it. Actually the more I dig into JWT, the more it looks like a nice solution since the token provider can still live outside LoRa Server, and the token itself can specify to which API methods and resources (e.g. which applications, nodes) the user has access to.

A token could be something like this:

```json
{
    "applications": ["0102030405060708"],
    "nodes": ["*"],
    "methods": ["Application.*", "Node.*", "NodeSession.*"],
    "exp": 12345678
}
```

brocaar, 2016-07-05:
I've been digging into adding auth. to the API. My plan is to replace the current API with gRPC*. This makes it possible to generate an API client (eventually clients for different programming languages) + gRPC has support for auth. metadata (the `net/rpc` package I'm currently using doesn't expose any request related metadata). Since gRPC doesn't have browser support (yet?) I'm planning to use https://github.com/gengo/grpc-gateway as a JSON proxy. See https://coreos.com/blog/gRPC-protobufs-swagger.html for an example. What do you think?

ivajloip, 2016-07-06:
I did some reading on the gRPC and it seems quite interesting :) I did not manage to understand however how the security is handled in the case of grpc-gateway. It seems to me that your intention is to use the metadata for authentication/authorization purposes. If I understood correctly, this means that headers that you send will be transformed into metadata. To set the headers, this means for me that the javascript code will have to know an access token. In other words, it will need to store it somewhere (it does not seem practical to receive it on every request). If you want to store it using javascript, you will need to store it in the web storage. That is not recommended by a lot of people as any javascript on your page (think any javascript dependency, which might get compromised) has access to it. 
On the other hand, you can use a combination of cookies + headers (seen by some people as the best option for storing tokens) for the authentication itself, but then I am not sure if a possible switch to gRPC is auth. related. If you switch to it you will still have advantages for clients in other languages and the use of protobufs.
I might be missing something, but basically those are the things I have some doubts about.

P.S: I have already implemented a solution for another project using only JWT and maybe this is clouding my judgement :)

brocaar, 2016-07-28:
@ivajloip The first step has been implemented and tagged as 0.9.0 :-) This new api makes it possible to implement authentication layers on top of it, something that the old implementation was lacking. I'll continue on the JWT auth. in the next release :-)

brocaar, 2016-08-10:
This has been implemented and released as `0.10.0`

closed 2016-08-10 by brocaar