brocaar/chirpstack-network-server #21: mqtt connection error

raphaelpeters, 2016-05-20:
When I start the semtech-bridge and loraserver both, I get following errors:

loraserver:
...
INFO[0001] application/mqttpubsub: connected to mqtt server 
INFO[0001] application/mqttpubsub: subscribing to tx topic  topic=application/+/node/+/tx
ERRO[0001] application/mqttpubsub: mqtt connection error: write tcp 127.0.0.1:54411->127.0.0.1:1883: use of closed network connection 
INFO[0001] application/mqttpubsub: connected to mqtt server 
INFO[0001] application/mqttpubsub: subscribing to tx topic  topic=application/+/node/+/tx
ERRO[0001] application/mqttpubsub: mqtt connection error: EOF 
INFO[0001] application/mqttpubsub: connected to mqtt server 
INFO[0001] application/mqttpubsub: subscribing to tx topic  topic=application/+/node/+/tx
ERRO[0001] application/mqttpubsub: mqtt connection error: EOF 
INFO[0001] application/mqttpubsub: connected to mqtt server 
INFO[0001] application/mqttpubsub: subscribing to tx topic  topic=application/+/node/+/tx
ERRO[0001] application/mqttpubsub: mqtt connection error: EOF 
INFO[0001] application/mqttpubsub: connected to mqtt server 
...

semtech-bridge:
...
ERRO[0007] backend/mqttpubsub: mqtt connection error: write tcp 127.0.0.1:54468->127.0.0.1:1883: write: broken pipe 
ERRO[0007] backend/mqttpubsub: mqtt connection error: EOF 
INFO[0007] backend/mqttpubsub: connected to mqtt server 
INFO[0007] Backend/mqttpubsub: re-registering to gateway topics  topic_count=1
ERRO[0007] backend/mqttpubsub: mqtt connection error: EOF 
INFO[0007] backend/mqttpubsub: connected to mqtt server 
INFO[0007] Backend/mqttpubsub: re-registering to gateway topics  topic_count=1
INFO[0007] backend/mqttpubsub: connected to mqtt server 
INFO[0007] Backend/mqttpubsub: re-registering to gateway topics  topic_count=1
ERRO[0007] backend/mqttpubsub: mqtt connection error: EOF 
INFO[0007] backend/mqttpubsub: connected to mqtt server 
INFO[0007] Backend/mqttpubsub: re-registering to gateway topics  topic_count=1
ERRO[0007] backend/mqttpubsub: mqtt connection error: EOF 
INFO[0007] backend/mqttpubsub: connected to mqtt server 
INFO[0007] Backend/mqttpubsub: re-registering to gateway topics  topic_count=1
...

I only get these mqtt packets:
gateway/b827ebfffee35329/rx {"rxInfo":{"mac":"b827ebfffee35329","time":"2016-05-20T12:57:52.481279Z","timestamp":111316044,"frequency":868300000,"channel":8,"rfChain":1,"crcStatus":1,"codeRate":"4/5","rssi":-16,"loRaSNR":9.5,"size":29,"dataRate":{"modulation":"LORA","spreadFactor":7,"bandwidth":250}},"phyPayload":"QJt7o82ABQACoYFmhN89oWNyYUotAX8qvBoFAR8="}

But no application packets.

brocaar, 2016-05-20:
Could you post your MQTT broker version and MQTT logs?

raphaelpeters, 2016-05-20:
1463751297: mosquitto version 0.15 (build date 2013-08-23 19:23:43+0000) starting
1463751297: Opening ipv4 listen socket on port 1883.
1463751297: Opening ipv6 listen socket on port 1883.
1463751300: New connection from 127.0.0.1.
1463751300: Invalid protocol "MQTT" in CONNECT from 127.0.0.1.
1463751300: Socket read error on client (null), disconnecting.
1463751300: New connection from 127.0.0.1.
1463751300: New client connected from 127.0.0.1 as .
1463751301: New connection from 127.0.0.1.
1463751301: Invalid protocol "MQTT" in CONNECT from 127.0.0.1.
1463751301: Socket read error on client (null), disconnecting.
1463751301: New connection from 127.0.0.1.
1463751301: Client  already connected, closing old connection.
1463751301: New client connected from 127.0.0.1 as .
1463751301: New connection from 127.0.0.1.
1463751301: Client  already connected, closing old connection.
1463751301: New client connected from 127.0.0.1 as .
1463751301: New connection from 127.0.0.1.
1463751301: Invalid protocol "MQTT" in CONNECT from 127.0.0.1.
1463751301: Socket read error on client (null), disconnecting.
1463751301: New connection from 127.0.0.1.
1463751301: Client  already connected, closing old connection.
1463751301: New client connected from 127.0.0.1 as .
1463751301: New connection from 127.0.0.1.
1463751301: Client  already connected, closing old connection.
1463751301: New client connected from 127.0.0.1 as .
1463751301: New connection from 127.0.0.1.
1463751301: New connection from 127.0.0.1.
1463751301: Client  already connected, closing old connection.
1463751301: New client connected from 127.0.0.1 as .
1463751301: New connection from 127.0.0.1.
1463751302: Client  already connected, closing old connection.
1463751302: New client connected from 127.0.0.1 as .
1463751302: New connection from 127.0.0.1.
1463751302: Client  already connected, closing old connection.
1463751302: New client connected from 127.0.0.1 as .
1463751302: Client  already connected, closing old connection.
1463751302: New client connected from 127.0.0.1 as .
1463751302: New connection from 127.0.0.1.
1463751302: Client  already connected, closing old connection.
1463751302: New client connected from 127.0.0.1 as .
1463751302: New connection from 127.0.0.1.
1463751302: Client  already connected, closing old connection.
1463751302: New client connected from 127.0.0.1 as .
1463751302: New connection from 127.0.0.1.
1463751302: Client  already connected, closing old connection.
1463751302: New client connected from 127.0.0.1 as .
1463751302: New connection from 127.0.0.1.
1463751302: New connection from 127.0.0.1.
1463751302: Client  already connected, closing old connection.
1463751302: New client connected from 127.0.0.1 as .
1463751302: Client  already connected, closing old connection.
1463751302: New client connected from 127.0.0.1 as .
1463751302: New connection from 127.0.0.1.
1463751302: Client  already connected, closing old connection.
1463751302: New client connected from 127.0.0.1 as .
1463751302: New connection from 127.0.0.1.
1463751302: New connection from 127.0.0.1.
1463751302: Client  already connected, closing old connection.
1463751302: New client connected from 127.0.0.1 as .
1463751302: Client  already connected, closing old connection.
1463751302: New client connected from 127.0.0.1 as .
1463751302: New connection from 127.0.0.1.
1463751302: New connection from 127.0.0.1.
1463751302: Client  already connected, closing old connection.
1463751302: New client connected from 127.0.0.1 as .
1463751302: Client  already connected, closing old connection.
1463751302: New client connected from 127.0.0.1 as .
1463751302: New connection from 127.0.0.1.
1463751302: Client  already connected, closing old connection.
[... 105 lines cut ...]
1463751302: New client connected from 127.0.0.1 as .
1463751302: New connection from 127.0.0.1.
1463751302: New connection from 127.0.0.1.
1463751302: Client  already connected, closing old connection.
1463751302: New client connected from 127.0.0.1 as .
1463751302: New connection from 127.0.0.1.
1463751302: Client  already connected, closing old connection.
1463751302: New client connected from 127.0.0.1 as .
1463751302: New connection from 127.0.0.1.
1463751302: Client  already connected, closing old connection.
1463751302: New client connected from 127.0.0.1 as .
1463751302: New connection from 127.0.0.1.
1463751302: Client  already connected, closing old connection.
1463751302: New client connected from 127.0.0.1 as .
1463751302: New connection from 127.0.0.1.
1463751302: Client  already connected, closing old connection.
1463751302: New client connected from 127.0.0.1 as .
1463751302: New connection from 127.0.0.1.
1463751302: Client  already connected, closing old connection.
1463751302: New client connected from 127.0.0.1 as .
1463751302: New connection from 127.0.0.1.
1463751302: Client  already connected, closing old connection.
1463751302: New client connected from 127.0.0.1 as .
1463751302: New connection from 127.0.0.1.
1463751302: Client  already connected, closing old connection.

raphaelpeters, 2016-05-20:
That's a very old version, I update manually and try again.

raphaelpeters, 2016-05-20:
Ok, it works now :')

closed 2016-05-20 by raphaelpeters