brocaar/chirpstack-network-server #22: Cannot send TXpacket from server to im880a

bu62963, 2016-05-28:
Hi!
I am new to LoRaWAN, trying to setup my own network using iM880a + iC880 + RaspberryPi + loraserver.
And I managed to send sensor data from iM880 to the server then applications.
However, it seems like I cannot send any packet from applications to the server then gateway then iM880.

[image]

It stops here, I am not sure where the problem is.

[image]
[image]
[image]

Is anyone willing to help me?

Thank you so much!

Kevin

brocaar, 2016-05-28:
Please note that the node must initiate a receive window by sending a
packet to the application. LoRa Server will then check if there are
downlink packets in the queue and send them back to the node.

I realise now that this is undocumented, but when the application receives
a payload from the node and replies directly with a downlink payload back
to LoRa Server, it will be transmitted during rx1 or rx2 (currently only
rx1).

See the LoRaWAN specs for more info about receive windows.

Hope this helps!

bu62963, 2016-05-28:
Thank you for replying me, brocaar.

I am using an example code in LMiC 1.5 on iM880a, and it sends a message to the server every 20s.

[image]

Is that means iM880a has initiated a receive window already?

[image]

All the messages from application are stored in the queue.

BTW, when I was testing the server with WiMOD LoRaWAN EndNode Studio, iM880a was able to join and send messages periodically, but could not receive any message from the server/application.

Thank you again! I will search on receive windows and try to understand it.

Kevin

brocaar, 2016-05-28:
Yes, your setup is correct 😄 

After every transmission the node opens two receive windows; rx1 and rx2. Do you see any message being sent to the gateway (``gateway/+/tx`` MQTT topic)? If so, it might be an issue at the gateway side / misconfiguration? Note that you can compile the ``basic_pkt_forwarder`` with debugging enabled, which might be helpful to debug issues at the gateway side. As well keep an eye on the [error topic](http://loraserver.readthedocs.io/en/latest/receiving-data/#applicationappeuinodedeveuierror).

bu62963, 2016-05-28:
Thank you again, brocaar.

However, the problem is that I cannot see any MQTT message sent to or from semtech bridge when I send a message from the application (Node-red) to the loraserver.

When iM880 is joining the network, I found this logged by loraserver:
[image]
And this on semtech bridge:
[image]

So the connection should be correct.

But, when I send a message from application to the loraserver, I found this:
[image]
And there is no follow-up.

I am confused...

Then I check all the MQTT messages, each time the iM880 sends a message, there is this:
[image]
and only this message, sent to the semtech bridge.

Is the message encoded into phpPayload?

Thank you for helping me!

Kevin

brocaar, 2016-05-28:
Yes, that is a downlink message to be transmitted by the gateway, probably with the payload sent by the application.

bu62963, 2016-05-28:
I see!

Thanks!

brocaar, 2016-06-06:
Could you let me know if your issue has been resolved?

bu62963, 2016-06-08:
No, sorry.
But I have changed to ttn server, and now everything is ok.

siscia, 2016-06-30:
@brocaar 

I would like to re-open this thread.

I tried to send a message via HTTP post request and everything I get is a 500 error code.

I tried to take a look into the code base but I wasn't able to find where the `/application` endpoint is handle.

From there onward I think I can manage by myself to fix my issue...

brocaar, 2016-06-30:
Note that you need to send your downlink payload over MQTT (JSON encoded): http://docs.loraserver.io/loraserver/sending-data/. `application/[AppEUI]/node/[DevEUI]/tx` is not an http endpoint but a MQTT topic :-)

arjansplit, 2016-07-04:
Hi Brocaar, i would like to follow up on the questions "bu62963" asked.

I am experiencing a similar problem with sending a downlink (application ->node) transmission.
When sending  my downlink payload over MQTT (JSON encoded), like you said in previous post i do get a confirm message form my loraserver that the "tx payload is received" and that the tx-payload is added to queue. In order to receive the downlink message I initiate an uplink transmission from my node so that the node can receive the downlink message which is in queue by the gateway. 
But after the uplink message has been received i get an error in the loraserver command line: "the processing rx packet error"

i have included a screenshot of the output of the loraserver commandline.
Do you have any idea what could be the problem and how i could solve this?

Thanks in advance, Arjan Split

[image]

brocaar, 2016-07-04:
I think I might know what the issue is in your case. Did you setup any custom channels? It looks like there is a bug in the code related to that. I'll look into it later today.

arjansplit, 2016-07-04:
I did setup some custom channels, I also discovered in the file "band_eu863_870.go" that only these channels occur :"868100000,868300000,868500000".

For my custom channel list i used all the frequencies that occur in the following file listed on: "https://github.com/Lora-net/packet_forwarder/blob/master/lora_pkt_fwd/global_conf.json"

summarized:
Channel # // Frequency
3 ; 867100000
4 ; 867300000
5 ; 867500000
6 ; 867700000
7 ; 867900000

brocaar, 2016-07-04:
The first three must always be implemented, the rest is user defined. I think that is where it goes wrong at the LoRa Server side (I'm doing some validation, but I think I've forgotten to consider custom channels). Thanks for reporting it 👍

brocaar, 2016-07-04:
@arjansplit I've confirmed the issue and fixed it (https://github.com/brocaar/lorawan/commit/16849c67e8e7b8d7896e8a5f05a650f953f69471 + some changes in loraserver itself to make use of this). The fixed version will be available as 0.8.1 shortly.

arjansplit, 2016-07-05:
Okay, i will download the new version,

Thank you very much!

arjansplit, 2016-07-05:
Hi Brocaar, I've downloaded the new 0.8.1 version and my problems are fixed. Thanks for that!

Something what does happen in the loraserver commandline is that it is giving the same error messages as before when receiving packets on the frequencies that are non-standard (i.e. channel 3-7 867.1 , 867.3, 867.5, 867.7, 867.9)
The received messages do appear in my MQTT broker, so i conclude that the messages are being forwarded correctly. I guess the error giving in the command line is false but I'm not sure.

Attached i have a screenshot of the Loraserver commandline with the given errors.
The messages where the error is not shown are received in the standard 868,1 868,3 868,5 band and don't trow the error.

So no big issue for me, but maybe you would like to know of the issue.

Greetings, Arjan

[image]

brocaar, 2016-07-05:
Could you try / validate the following:

* Did you setup the custom channel-list in the web-interface / API
* Did you assign this channel-list to the node in the web-interface?
* Could you do a new OTAA and test again? This will provision your node with the channel-list and will update the node-session. Since it stores the `CFList` within the node-session, it might still be empty when the session was created by LoRa Server 0.8.0.

arjansplit, 2016-07-05:
1. I did setup a channel list via the web-interface.
2. I did assign the custom channel list via the web-interface to the node.
3. I used ABP for activating my nodes. I made a whole new application with all new identification keys. I've changed the DevAddr, AppUEI, DevEUI, Appkey, AppSKey and NwkSKey to new value's.
Is it necessary to use OTAA to use the custom channel list? My node wouldt respond like i wanted when using OTAA so i used ABP instead for quick debugging. I could retry to get OTAA working.

Screens attached of the channel-list and channel list assignment to the node:
[image]
[image]
[image]

arjansplit, 2016-07-05:
Hi Brocaar,
I've managed to get OTAA working on my node.
The error message doesn't appear anymore so you could dismiss my previous questions regarding my previous question.

Thanks for your effort so far!

brocaar, 2016-07-05:
Great! Technically it is possible to to APB with extra channels, but when creating the node-session these channels must be given. This is not yet possible through the web-interface, but can already be done through the API:

```json
{
    "method": "NodeSession.Create",
    "params": [
        {
            "devAddr": "00000000",
            "appEUI": "0000000000000000",
            "devEUI": "0000000000000000",
            "appSKey": "00000000000000000000000000000000",
            "nwkSKey": "00000000000000000000000000000000",
            "fCntUp": 0,
            "fCntDown": 0,
            "rxDelay": 0,
            "rx1DROffset": 0,
            "cFlist": [channel4,channel5,channel6,channel7,channel8]
        }
    ],
    "id": 1
}
```
(replacing `channelx` with the channel frequency) See also: https://docs.loraserver.io/loraserver/api/

closed 2016-07-05 by brocaar

rohitkumar5293, 2017-05-15:
Hi brocaar,
I have implemented single packet forwarder only uplink and unable to do downlink. I want to send packet from gateway to node any message from gateway to end node. How this going to be possible with only uplink. I mean when my gateway send packet to server after that gateway should able to send some packet to end node at the same time. Is this possible? And if it is possible how can i implement? please help me out

Regards
Rohit

commit 74ce9c62 (2016-07-04): Fix 'unknown channel for frequency' error.

Closes #22.