brocaar/chirpstack-network-server #23: Multiple end-nodes join issue

Mehradzie, 2016-06-02:
Hi guys,

After whole lot of wrestling with bits and peices of LoRa today I was trying to connect a 2nd end-node to LoraServer, and surprisingly, when I have one end-node already joined, the 2nd one refuses to join.

I have tested them with the Loraserver which comes with Multitech AEP Conduit and they connected fine, so in this instance either Multitech mDot end-node library and Multitech Conduit gateway both have the same bug which enables me to connect multiple end-nodes or something isn't set up in the brocaar/loraserver right.

Tomorrow I am going to try flashing the Multitech mDots with Semtech open source LoraWAN firmware to see if I'll have the same issue but meanwhile I thought, I'll ask this and you'll get it see it tonight.

atm, on the Loraserver GUI I have set up a single Application (0000000000000001) and my end-nodes have two different Appkeys (00000000000000000000000000000001 and 00000000000000000000000000000002) and each one individually connects to the server alright but not together.

let me now what you think about this,

Cheers

Mehrad

Mehradzie, 2016-06-02:
[Update]
LoRaWAN semtech code for end-nodes seems to have a similar issue. One node hugs the server and non of the others can join. I am not even sure where to start to bug-fix this.

brocaar, 2016-06-02:
Hmmm... that is odd. I've beed doing tests in the past with multiple nodes, but maybe a regression happend during some refactoring or changes? I'll do some testing at my side to see if I can reproduce this issue and find a fix.

VirTERM, 2016-06-02:
I don't have such problem. Running multiple nodes through more than one gateway

Mehradzie, 2016-06-03:
I've put a 10s delay before sending the packets after the join and now the rest of the nodes (3 at the same time) join, no problem. gonna cram in another 10 to see how it will handle it. How frequently you guys have been sending packets for your testing?

I am guessing we are still dealing with the sx1310 single buffer issue with the current basic/gps/etc packet forwarder which doesn't have any queue system and packets are writing on top of each-other. :space_invader:

VirTERM, 2016-06-03:
For testing i send packet every 2
mins. Usually, multiple gateways and couple of nodes per gateway.

ivajloip, 2016-06-03:
I have also tested with multiple devices without a problem. In my case there was a large time difference between the join requests and thus the buffer size was not an issue for me.

Mehradzie, 2016-06-06:
Thanks guys. I have figured out the during the 5 second wait for the join response, the request was getting over written by another node's 2 second data. Hope they implement the queue on the stable edition of packet forwarder soon.

closed 2016-06-06 by Mehradzie

VirTERM, 2016-06-20:
Actually, I run some tests, and i can see that there is a problem. If i send even a couple of join requests via the same gateway, at least one of them will fail....

brocaar, 2016-06-20:
Did you already try the new packet_forwarder (v3) with the
lora-semtech-bridge (v2)? The new packet_forwarder should fix some issues
regarding to scheduling.