brocaar/chirpstack-network-server #24: DevAddr problem on Rx

VirTERM, 2016-06-02:
In my test setup, I have couple of nodes sending payload to the server. There is no problem with joining, however one of the nodes's DevAddr seems to be read incorrectly:

INFO[2753] gateway: sending udp packet to gateway        addr=x.x.x.x54847 type=PushACK
INFO[2754] gateway: received udp packet from gateway     addr=y.y.y.y:56394 type=PushData
INFO[2754] gateway: rxpk packet received                 addr=y.y.y.y:56394 data=QEFBMxwAAAACzhgt2HnzjwnH0GAuqqhyCOC0g1QR2mCVyR5Lfo6AhQ== mac=0080000000009zzz
INFO[2754] backend/mqttpubsub: publishing packet         topic=gateway/0080000000009bbb/rx
INFO[2754] gateway: sending udp packet to gateway        addr=y.y.y.y:56394 type=PushACK
INFO[2754] gateway/mqttpubsub: rx packet received       
ERRO[2754] processing rx packet error: get node-session for **DevAddr 1c334141** error: redigo: nil returned  data_base64=QEFBMxwAAAACzhgt2HnzjwnH0GAuqqhyCOC0g1QR2mCVyR5Lfo6AhQ==

Under Session/ABP in the server I can see different DevAddr associated with that node (**630e8d31**) and the packet counter obviously doesn't increment

VirTERM, 2016-06-02:
This problem occurs with nodes sending payload to multiple servers, I will further investigate and follow up if necessary, closing for now.

closed 2016-06-02 by VirTERM

mzemb, 2016-12-26:
Hi @VirTERM
Can you elaborate a little bit ? I have the same issue (error message at least), but I don't understand what you mean by "sending payload to multiple servers". Are you talking about the configuration of the gateway or of the gateway-bridge ?