brocaar/chirpstack-network-server #27: Idle State! Is there such a thing in loraserver/semtech-bridge?

Mehradzie, 2016-06-09:
Hey guys,

This morning, after I came back to my setup, I have notice that my server is inactive and since I was already tailing the logs I had a quick look and fair enough the last message on the Semtech-Bridge logs was,

`level=info msg="backend/mqttpubsub: unsubscribing from topic" topic="gateway/0000024b000e0000/tx"`

I have disabled all the testing end-nodes before I left, last night, so nothing was transmitting but I am wondering, if there is such a situation, would the server go to some sort of **Idle State** and snooze? And if so, how would you wake it up?

I have poked my gateway by restarting it's packet forwarder and the stream started flowing again but surely this can't be the solution.

I will do another follow up test tonight and report tomorrow,

Mehrad

ivajloip, 2016-06-09:
Hi,

Something like that happened a few days ago to me as well, but I did not have the time to take a closer look into it. I will try to reproduce it today/tomorrow and hopefully if there is a problem, we will manage to find it :)

brocaar, 2016-06-09:
This ``unsubscribing from topic`` happens when there isn't any activity from the gateway anymore after a given time, meaning your gateway / packet_forwarder doesn't send any stats packets (normally this is every 30/60 sec or so?). It shouldn't unsubscribe when the gateway is active, but there are no active nodes.

ivajloip, 2016-06-09:
In my case it is possible that the gateway was not sending packets. As far as I remember, a simple semtech-bridge restart did not solve the problem and I had to restart the packet forwarder as well.

Mehradzie, 2016-06-10:
Well, Murphy's law proves itself one more time just in case. I have left the setup in idle mode with removing all the end-nodes last night and guess what!! Nothing happened. When I came back to it this morning the gateway was still sending status reports and it was all good. 

I am assuming this is more of a gateway and packet forwarder issue, and when the packet forwarder is back online, the bridge and loraserver recovers and gets back to working condition with no intervention. 

I will close this topic until the bridge proven guilty. Thanks for the testing guys.

closed 2016-06-10 by Mehradzie