brocaar/chirpstack-network-server #28: The method to flush the unreceived data queue

Mehradzie, 2016-06-10:
Hi all,

I am wondering if there is a method to flush or shrink the un-received confirmed data queue.
While doing testing I echo heaps of data back to the end-nodes which based on the conditions they might not be received. Later on, when the issue is fixed, I receive heaps of old data buffered in the loraserver queue until I get to the bottom of the list and start receiving back, what I actually want to receive and I want some method of flushing that queue.

e.g.

```
end-node tx  --------- server tx (echo back)
   foo                            failed to echo back
   foo                            failed to echo back
   foo                            failed to echo back
   baz                           foo
   baz                           foo
   baz                           foo
   baz                           baz
   baz                           baz
   baz                           baz
```

ivajloip, 2016-06-10:
For me the following thing helps

1. Find the IP address of the redis `docker inspect loraserver_redis_1 | grep '"IPAddress"' | head -n 1 | tr -s " " | cut -f 3 -d " " | cut -f 2 -d '"'`
2. Connect to this redis instance `redis-cli -h <ip address>`
3. Remove the key that you need `del node_tx_queue_<devEUI>`

brocaar, 2016-06-10:
Will look into adding queue related methods to the API after I've finished the MAC command implementation.

Mehradzie, 2016-07-14:
That would be great. ATM, I am lazy and the moment I hit the `redis-cli` I issue
`FLUSHALL` which is a little bit harsh. :smile: 

*Feature Request*: Would also be great to be able to change the queue size externally as well. 
*Feature Request*: Could we have a *Feature Request* label please? :smiley:

commit 67090734 (2016-07-17): Implement FlushTXPayloadQueue API method.

Closes #28.
closed 2016-07-17 by brocaar