brocaar/chirpstack-network-server #30: LinkCheckAns

athtest800, 2016-07-02:
Hi!
I am new to lorawan and i have a device that makes LinkCheckReq. Is it possible to implement it with LinkCheckAns on the loraserver for automatic reply-answer to the device?

brocaar, 2016-07-02:
This part belongs to the responsibility of a network-controller, which is (currently) not part of LoRa Server. See also: http://docs.loraserver.io/loraserver/network-controller/. Keeping the network-controller part outside of LoRa Server enables you to add your own business logic to it :-)

athtest800, 2016-07-02:
i use node-red, mqtt output with this topic 
`application/0000000000000000/node/0102030405060708/mac/tx`
and this function
`var dd = {"reference": "abcd1234", 
    "devEUI": "0102030405060708",
    "frmPayload": false, 
    "macCommand": "Ag==" , //is this write for LinkCheckAns (0x02)??
    "data": "//8=" //here i use pre encoded base64 ??
    }
return {
   payload:dd
}`

But i get 
`"unmarshal mac command error: lorawan: invalid CID 2e" dev_eui=0102030405060708 reference=abcd1234`

Thank you

brocaar, 2016-07-03:
There is no ``data`` field: http://docs.loraserver.io/loraserver/network-controller/#sending-data

The full mac-command (including CID) needs to be base64 encoded and put into ``macCommand``

athtest800, 2016-07-03:
Thank you! Fixed

closed 2016-07-03 by athtest800

sopach, 2016-09-15:
Hi, is it possible send MAC command in FOpts together with message in FRMPayload ?