brocaar/chirpstack-network-server #31: support Class B

flexiot, 2016-07-02:
does the lora server support loraWAN class B?  lora gateway from semtech support beacon mode. it would be cool that the lora server support Class B including align the beacon rx slot, send frame at the right rx slot.

brocaar, 2016-07-02:
It doesn't currently but I'm planning to add class b and c support in the future.

yzk0281, 2016-09-12:
Hi, @brocaar 
How is this feature supporting class B now?
urgent to know~

gillespilloudwyres, 2016-10-22:
I'm interrested by this feature toi. Especially by the beaconning useful to synchronize devices.

brocaar, 2016-11-13:
@yzk0281 @gillespilloudwyres sorry for my late response, I was on vacation. Before my vacation I've been looking at class-b support. As the specs are still experimental and unstable and are likely to change in the LoRaWAN v1.1, I think it is better to wait a bit (during my testing I found out that the beacon format of the packet_forwarder was different than how the lmic firmware was expecting it).

wero1414, 2017-07-25:
Hey @brocaar  any news about this? 
I've read that the LoRa packet forwarder supports it, i don't know if it helps

[image]

brocaar, 2017-07-25:
The 1.1 specification itself is not out yet. I think it makes sense to wait for that and for nodes supporting the 1.1 specification.

gillespilloudwyres, 2017-07-25:
As I know, the class B transmission from semtech packet forwarder is only sent if a gps clock exist, and a gps fix is ok from the gateway. If you have only one gateway under your loraserver and your gateway is outside (can fix the gps), it will be ok. If you have two or more gateways, the the beacon will be sent by each gateways at the same time and the message will probably unreadable. Moreover, most of the gateways have not a gps embedded.

For me, the emission of the class B must be managed by the backend, with an algorithm to randomize wich gateway will do the emission of the class b beacon knowing that the devices need to tolerate 6 or 7 missing class b messages, to allow non-gps embedded into the gateways.

Bout you're right @brocaar we need to wait the 1.1 specification, even if others operators have already implemented the class b.

brocaar, 2017-07-25:
I'm currently working on channel management of the gateways. The first version will be implemented as a kind of sidekick service next to the packet-forwarder. On channel config changes, it will update the packet-forwarder JSON config and issues a packet-forwarder restart command. This could if needed, in the future be extended to manage which gateway will be responsible for beaconing. See also https://github.com/brocaar/lora-gateway-config (this has not yet been released officially).

brocaar, 2018-02-28:
Class-B support has been merged to the `master` branch :-) Will be released soon! I've been testing this successfully against https://github.com/lora-net/LoRaMac-node/tree/feature/class_b. 

This will follow the LoRaWAN 1.0.3 specification (I think this protocol version is back porting some Class-B logic from LoRaWAN 1.1: https://github.com/Lora-net/LoRaMac-node/issues/409).

closed 2018-02-28 by brocaar