brocaar/chirpstack-network-server #33: FATA[0000] NetID parse error: lorawan: exactly 3 bytes are expected

comet-bg, 2016-07-04:
Аt startup I get the following error:
FATA[0000] NetID parse error: lorawan: exactly 3 bytes are expected

/etc/systemd/system/loraserver.service.d/loraserver.conf content:

[Service]
Environment="NET_ID=010203"
Environment="BAND=EU_863_870"
Environment="HTTP_BIND=0.0.0.0:8000"
Environment="POSTGRES_DSN=postgres://loraserver:dbpassword@localhost/loraserver$

Environment="DB_AUTOMIGRATE=True"

Host platform: Raspberry PI3

brocaar, 2016-07-04:
Did you start the service with systemd or from the command-line?

comet-bg, 2016-07-04:
yes, I followed the getting started instructions:
pi@raspberrypi:~ $ loraserver
FATA[0000] NetID parse error: lorawan: exactly 3 bytes are expected
pi@raspberrypi:~ $

brocaar, 2016-07-04:
https://docs.loraserver.io/loraserver/getting-started/#starting-lora-server

```bash
# start
sudo systemctl start loraserver

# restart
sudo systemctl restart loraserver

# stop
sudo systemctl stop loraserver
```

comet-bg, 2016-07-04:
pi@raspberrypi:~ $ sudo systemctl start loraserver
pi@raspberrypi:~ $ journalctl -u loraserver -f -n 50
No journal files were found.

brocaar, 2016-07-04:
I'm not sure if I can help any further here, as every system is configured differently, but make sure you've create the `/etc/systemd/system/loraserver.service` file. As well, the systemd setup was to get you started, if that doesn't work you can also start loraserver from the cli with `loraserver`, passing the configuration as cli parameters like any other cli tool (see `loraserver --help`).

comet-bg, 2016-07-04:
thanks for the help!
strange but also for manual startup I get an error:
pi@raspberrypi:~ $ loraserver --net-id / 010203
FATA[0000] NetID parse error: encoding/hex: odd length hex string

brocaar, 2016-07-04:
Try it without the ``/`` 😉

comet-bg, 2016-07-04:
it helped :)
Apparently I have a problem with loraserver.conf file ...
Thank you!

closed 2016-07-04 by brocaar

wolftrax5, 2016-09-10:
hi @comet-bg  what was the error on loraserver.conf file? ; 
because i have the same problem but i can't fixed with the other tips 
[image]

brocaar, 2016-09-10:
@wolftrax5 note that when you start `loraserver` from the commandline directly (thus not using systemd), the systemd (`/etc/systemd/system/loraserver.service.d/loraserver.conf`) configuration won't be applied.

bouazizihsan, 2017-04-17:
Hi,I downloaded loraserver from this link:
Https://github.com/brocaar/loraserver/
Releases
But I have a problem when I type this command:
 # Sudo systemctl start loraserver
Failed to start loraserver.service: Unit loraserver.service not found.
Is what i doid to do the server vreation from the beginning parsque i downloaded only the binary file

greggjaskiewicz, 2017-08-22:
I'm getting similar issue, which appears in logs as:

```
Aug 22 09:00:45 ubuntu-xenial loraserver[25743]: time="2017-08-22T09:00:45Z" level=info msg="starting gateway api server" bind="0.0.0.0:8002" ca-cert= tls-cert= tls-key=
Aug 22 09:00:45 ubuntu-xenial loraserver[25743]: time="2017-08-22T09:00:45Z" level=fatal msg="--gw-server-jwt-secret must be set"
Aug 22 09:00:45 ubuntu-xenial systemd[1]: loraserver.service: Main process exited, code=exited, status=1/FAILURE
Aug 22 09:00:45 ubuntu-xenial systemd[1]: loraserver.service: Unit entered failed state.
Aug 22 09:00:45 ubuntu-xenial systemd[1]: loraserver.service: Failed with result 'exit-code'.
Aug 22 09:00:45 ubuntu-xenial systemd[1]: loraserver.service: Service hold-off time over, scheduling restart.
Aug 22 09:00:45 ubuntu-xenial systemd[1]: Stopped LoRa Server.
Aug 22 09:00:45 ubuntu-xenial systemd[1]: Started LoRa Server.
```

brocaar, 2017-08-22:
Hint 😉 :

> level=fatal msg="--gw-server-jwt-secret must be set"

greggjaskiewicz, 2017-08-22:
Yes, all I did - was upgrade the debs on the ubuntu installation. I also had to change the listening address to 1443, because loraserver binary, not being ran as root - cannot bind to the port 443

l0ng09, 2017-12-21:
Hi ,@brocaar I meet the same error,when I try to compile from the source 
make build 
cd build 
./loraserver
FATA[0000] NetID parse error: lorawan: exactly 3 bytes are expected 

when I run make test 
 Given a clean database and api instance ERRO[0000] ping database error, will retry in 2s: pq: password authentication failed for user "ryan" 
ERRO[0002] ping database error, will retry in 2s: pq: password authentication failed for user "ryan" 
ERRO[0004] ping database error, will retry in 2s: pq: password authentication failed for user "ryan" 
...

brocaar, 2017-12-21:
@l0ng09 I think the message is in the error logs 😉 Anyway, please move this discussion to https://forum.loraserver.io