brocaar/chirpstack-network-server #38: FCnt if node reset

athtest800, 2016-07-09:
Hi!
When the node reboot, i get not rx messages. How to avoid this issue? In the backend i have FCnt >0 (eg.50). When i reset it to 0, messages arrive. Should i manually change this or the backend can ignore it?
Thank you egain for this great software!

brocaar, 2016-07-17:
I think the best would be either to store the state of your node (so the FCnt is not reset), or perform a new OTAA on reset, which will assign a new session to the node (with FCnt=0).

spinarr, 2016-07-19:
http://cocoon.actility.com/system/files/wireless/ThingPark_Wireless%20PHY_and_MAC_layer_specifications_v1.pdf

Page 14, Frame counter field, Send only conditions

May be relevant to this issue?

xPrcek, 2016-08-19:
Loraserver compares fcnt from node with fcnt from server. And if node fcnt is smaller than server (but not vice versa), then ERRO[0249] occurs. Unfortunately, we have no info about this situation from message bus. The only way is inspect loraserver logs for this:

```
WARN[0249] invalid FCnt                                  dev_addr=055db212 dev_eui=12c5daaafc5e22f8 packet_fcnt=116 server_fcnt=340

ERRO[0249] processing rx packet error: invalid FCnt or too many dropped frames  data_base64=QDGz7QaAdAAIjIk6mxTVXKE=
```

Storing the state of node inside the node is not right way, because payload from node might not come to loraserver, so fcnt on the node side will be +1, but  on the server side will be the same (however this will not occurs ERRO[0249] , because node fcnt < server is ok). And of course, we don't like overwriting node flash.

I am not sure if the right way is that network server adopts the fcnt from node (because of LoRaWAN specification 1.0), but I am pretty sure, that we need a message bus info if this (ERRO[0249] ) situation occurs.

brocaar, 2016-08-19:
Please note that `ERRO[0249]` is not an error code, but it means that the error occurred 249 seconds after you started LoRa Server 😉 

Note that it is perfectly fine for the FCnt check when some frames get dropped (so never are received and / or processed by LoRa Server), as long is it doesn't exceed `MAX_FCNT_GAP`. However, when a node resets resulting in a `FCnt=0`, this is a problem, since LoRa Server can't validate what happened, it is either the node that was reset, or somebody else who is re-transmitting your first frame (and with that is able to reset the network-server counters).

I've seen some implementations that offer a "relaxt" FCnt validation, but I think when enabling that option, you sacrifice security. I would say that the best solution is that a node re-joins when it was reset. That way, the counters at the network-server can be reset in a secure way.

The good news is that I think this will be solved in the LoRaWAN 1.1 specs.

xPrcek, 2016-08-20:
Well, I believe we can handle this. Please note, for production, it is necessary to have info from message bus about this event with dev_eui, packet_fcnt and server_fcnt. Could you update the todo list, please?

xPrcek, 2016-08-21:
Feedback from node hardware developers: Storing state of fcount on the flash disk will consume undesirable more power and shrink flash disk. The "OTAA-way reset" is kind of dirty workraound for ABP. Unfortunately, we can't wait for LoRaWAN spec 1.1, any other ideas?

kruisdraad, 2016-09-28:
For testing it would be really handy to have the 'relax frame counter' option to be enabled to ignore the frame count

brocaar, 2016-10-05:
As it stays a popular question, I'll implement an optional relaxt frame-counter mode :-)

The LoRa Server part has already been implemented in https://github.com/brocaar/loraserver/commit/3333fbb97f5ef796eb423b24f53a9bf081f5c9ca. The ui part is still todo.

closed 2016-10-05 by brocaar

brocaar, 2016-10-05:
Please upgrade to LoRa Server 0.12.3 and LoRa App Server 0.1.2 and you'll see the relax frame-counter option in the node form :-)

[image]

gousmodin, 2017-10-28:
Hi,

Even i have set relax frame counter, and reset the node , the issue of too many dropped frames persists. Any solution is appreciated. I am struck. 

Fyi,
I have node with confirmef msg on. And lora server is not able to send downlink message to node due to the issue.

commit d3bf3b1d (2016-10-05): Implement relax frame-count.

This optional option resets the FCntUp and FCntDown to 0 after a (possible) node reset (or replay-attack).

This closes #38.