brocaar/chirpstack-network-server #39: Hi, I setup Loraserver but when I try to visit the http page got nothing

yzk0281, 2016-07-12:
./loraserver --net-id 010203 --band EU_863_870 --http-bind 0.0.0.0:8000 --postgres-dsn postgres://loraserver:dbpassword@localhost/loraserver?sslmode=disable --redis-url redis://localhost:6479 --db-automigrate
I start loraserver with the commend above from the precompiled binary file
All things seem work well according to log:
time="2016-07-12T14:30:10+08:00" level=info msg="starting LoRa Server" band="EU_863_870" net_id=010203 version=0.8.1 
time="2016-07-12T14:30:10+08:00" level=info msg="connecting to postgresql" 
time="2016-07-12T14:30:10+08:00" level=info msg="setup redis connection pool" 
time="2016-07-12T14:30:10+08:00" level=info msg="gateway/mqttpubsub: connecting to mqtt broker" server="tcp://localhost:1883" 
time="2016-07-12T14:30:10+08:00" level=info msg="application/mqttpubsub: connecting to mqtt broker" server="tcp://localhost:1883" 
time="2016-07-12T14:30:10+08:00" level=info msg="gateway/mqttpubsub: connected to mqtt server" 
time="2016-07-12T14:30:10+08:00" level=info msg="gateway/mqttpubsub: subscribing to rx topic" topic="gateway/+/rx" 
time="2016-07-12T14:30:10+08:00" level=info msg="controller/mqttpubsub: connecting to mqtt broker" server="tcp://localhost:1883" 
time="2016-07-12T14:30:10+08:00" level=info msg="application/mqttpubsub: connected to mqtt server" 
time="2016-07-12T14:30:10+08:00" level=info msg="application/mqttpubsub: subscribing to tx topic" topic="application/+/node/+/tx" 
time="2016-07-12T14:30:10+08:00" level=info msg="applying database migrations" 
time="2016-07-12T14:30:10+08:00" level=info msg="controller/mqttpubsub: connected to mqtt broker" 
time="2016-07-12T14:30:10+08:00" level=info msg="controller/mqttpubsub: subscribing to tx topic" topic="application/+/node/+/mac/tx" 
time="2016-07-12T14:30:10+08:00" level=info msg="migrations applied" count=0 
time="2016-07-12T14:30:10+08:00" level=info msg="registering json-rpc handler" path="/rpc" 
time="2016-07-12T14:30:10+08:00" level=info msg="registering gui handler" path="/" 
time="2016-07-12T14:30:10+08:00" level=info msg="starting http server" bind="0.0.0.0:8001" 

but when I visit http://ip:8001, I got a blank page(only with a title'LoRa Server'),  anyone has suggestions?

Mehradzie, 2016-07-12:
If you mean to see the GUI for the Loraserver, try port `3000`

yzk0281, 2016-07-12:
@Mehradzie  thanks
but port 3000 does not work

yzk0281, 2016-07-12:
I want to modify nodes

yzk0281, 2016-07-12:
Should I put the static folder somewhere?
I only have a pre-compiled binary file 'loraserver' now~

Mehradzie, 2016-07-12:
I am sorry. I have no idea why I thought that is port `3000`. I have it open on front of me on `8000`. Must be getting old :)

I must be dreaming again but from you said above
> `./loraserver --net-id 010203 --band EU_863_870 --http-bind 0.0.0.0:8000`
so you chose your http server to be on 8000, however, the loraserver log shows that it's trying to open port `8001` instead :O 
>`time="2016-07-12T14:30:10+08:00" level=info msg="starting http server" bind="0.0.0.0:8001"`

Could you try the the link below to kinda jump to the applications page, in case there is a problem with loading the main page? 
`<your ip address>:8000/#/applications`

yzk0281, 2016-07-12:
blank still

yzk0281, 2016-07-12:
any way to see the http log of loraserver?

yzk0281, 2016-07-12:
quite confused

Mehradzie, 2016-07-12:
Could you provide an screenshot?
>but when I visit http://ip:8001, I got a blank page(only with a title'LoRa Server'),

yzk0281, 2016-07-12:
[image]

yzk0281, 2016-07-12:
just like this

Mehradzie, 2016-07-12:
Is *loraserver* running on a VM? if so, have you given permissions to port `8000` , `8001` or any port that you chose to use to be accessed from outside world? :)

Mehradzie, 2016-07-12:
Other than than, I am sorry @yzk0281. I don't have any other ideas in mind. I have never had a problem accessing that page in the past. Let's see what others think about this.

yzk0281, 2016-07-12:
@Mehradzie  thank you~
this is a strange problem~

yzk0281, 2016-07-12:
or I got wrong go version?
@Mehradzie which version do you use?

Mehradzie, 2016-07-12:
Just get the fully compiled loraserver from the *releases* tab in the loraserver github page and do a testing with that first.

yzk0281, 2016-07-12:
I mean the go version

yzk0281, 2016-07-12:
seems to be the problem of ‘https://ajax.googleapis.com’， can not  access in China~
thanks
@Mehradzie

closed 2016-07-12 by yzk0281

aminatamima, 2016-12-15:
Hi @yzk0281, how did you solve this problem please? The link given in your precedent comment does not work anymore.