brocaar/chirpstack-network-server #40: Try to build LoraServer but error occured

yzk0281, 2016-07-13:
go get github.com/brocaar/loraserver/...
then use commend 'make build' to build from the source code but error occur:
# github.com/jmoiron/sqlx
../../jmoiron/sqlx/bind.go:142: undefined: strings.IndexByte
../../jmoiron/sqlx/sqlx.go:603: db.Ping undefined (type *DB has no field or method Ping)
# github.com/garyburd/redigo/redis
../../garyburd/redigo/redis/conn.go:97: undefined: net.Dialer
# github.com/Sirupsen/logrus
../../Sirupsen/logrus/writer.go:19: method logger.Debug is not an expression, must be called
../../Sirupsen/logrus/writer.go:21: method logger.Info is not an expression, must be called
../../Sirupsen/logrus/writer.go:23: method logger.Warn is not an expression, must be called
../../Sirupsen/logrus/writer.go:25: method logger.Error is not an expression, must be called
../../Sirupsen/logrus/writer.go:27: method logger.Fatal is not an expression, must be called
../../Sirupsen/logrus/writer.go:29: method logger.Panic is not an expression, must be called
../../Sirupsen/logrus/writer.go:31: method logger.Print is not an expression, must be called
../../Sirupsen/logrus/writer.go:41: undefined: bufio.NewScanner
# github.com/brocaar/lorawan
../lorawan/phypayload.go:512: function ends without a return statement
# github.com/eclipse/paho.mqtt.golang/packets
../../eclipse/paho.mqtt.golang/packets/packets.go:224: function ends without a return statement
# github.com/elazarl/go-bindata-assetfs
../../elazarl/go-bindata-assetfs/assetfs.go:141: function ends without a return statement
# github.com/rubenv/sql-migrate/sqlparse
../../rubenv/sql-migrate/sqlparse/sqlparse.go:19: undefined: bufio.NewScanner
../../rubenv/sql-migrate/sqlparse/sqlparse.go:20: undefined: bufio.ScanWords
../../rubenv/sql-migrate/sqlparse/sqlparse.go:49: undefined: bufio.NewScanner
# github.com/lib/pq
../../lib/pq/conn.go:176: undefined: bufio.NewScanner
# github.com/codegangsta/cli
../../codegangsta/cli/context.go:335: function ends without a return statement
make: *** [build] Error 2

anyone has any suggestion？

Mehradzie, 2016-07-13:
What OS you're issuing the `make build` command?
You can issue a `uname -a` and post the response.

yzk0281, 2016-07-13:
Linux iZ23ny5ud5oZ 3.8.0-44-generic #66~precise1-Ubuntu SMP Tue Jul 15 04:01:04 UTC 2014 x86_64 x86_64 x86_64 GNU/Linux
ubuntu 14.04 server

Mehradzie, 2016-07-13:
and the response for 
`go version`

yzk0281, 2016-07-13:
go version go1

Mehradzie, 2016-07-13:
Say, from your first error that you're receiving 

> ../../jmoiron/sqlx/bind.go:142: undefined: strings.IndexByte

The function strings.IndexByte was added in go version 1.2. So the rest of your issues can be related to that as well. 

Update your go to the latest and try again. I am using **go1.6** and everything looks to be fine.

yzk0281, 2016-07-13:
thank you very much~~~

closed 2016-07-13 by yzk0281