brocaar/chirpstack-network-server #42: How to add ISM band? eg.470-478?

yzk0281, 2016-07-13:
How to add ISM band? eg.470-478?

ivajloip, 2016-07-13:
You need to add a file [here][1] (get inspiration from the already available options) for your band and modify [this][2] in order to be able to use your new ISM band.

[1]: https://github.com/brocaar/lorawan/tree/master/band
[2]: https://github.com/brocaar/lorawan/blob/master/band/band.go#L180

Mehradzie, 2016-07-14:
@yzk0281 from your previous comment I am assuming you're located in China. I appologise in advance if I'm wrong. 
> seems to be the problem of ‘https://ajax.googleapis.com’， can not access in China~

Based on that assumption, there is defined band for **China** in the **LoraWAN** standard published by *LoRa Alliance*, for `779-787MHz` **ISM Band**. (LoRaWAN 1.0 Specification. Section 7.3)

You can request your own copy of the specification from [this link](https://www.lora-alliance.org/Contact/Request-Specification-Form). And then as @ivajloip pointed out, you can start by copying one of the existing bands (probably the closest one for your ISM band) and start editing it. If you have any problems doing so, I can give you hand if I can open up some free time. :smiley:

yzk0281, 2016-07-14:
@ivajloip  thanks for answering~
@Mehradzie band range here in China is 470 - 478
I will appreciate that if you could give a hand with this headache problem

brocaar, 2016-07-17:
@yzk0281 I'm more than happy to help you out with implementing a China band. Unfortunately I'm unable to test it (which is the reason why it isn't implemented yet), but it would be great if we can work together on this.

yzk0281, 2016-07-17:
@brocaar  I can test it, I got nodes and gateways currently working on 470 -478 but connecting to loraiot
I'm so glad that you can help me with that~
If you add it to the master, I will test it as soon as possible

brocaar, 2016-07-17:
I've created branch named `test_cn470_510` with a test implementation of the CN 470-510 ISM band. After compiling, you should be able to run `loraserver` with `--band CN_470_510` Looking forward to your test results!

mkurnadi, 2016-07-22:
Hi @brocaar could you provide 433 MHz ISM band as well for rest of the world ?

brocaar, 2016-07-22:
Yes, no problem. Could you please create a separate github issue for this?
And could you let me know if you're able to help out with testing (I don't
have any hardware for this band to test with).

mkurnadi, 2016-07-23:
Sure, I already have devices to test at 433 MHz. I just opened a separate github issue for this.

brocaar, 2016-07-23:
@yzk0281 could you please let me know if the following works for you?

- [ ] data up (node -> loraserver)
- [ ] data down (loraserver -> node)
- [ ] OTAA (over-the-air activation)

I'll merge these changes into `master` as soon as I have the confirmation from you (or somebody else who is able to test this ISM band) and tag a new release :-)

panyingyun, 2016-07-25:
@yzk0281  could you tell me your  email or qq,  Email: panyingyun(at)gmail.com   QQ:919224283. our team also test 470~510 mode, so we can help each other

brocaar, 2016-07-25:
@panyingyun @yzk0281 if you want, I can provide pre-compiled binaries to test with for the 470-510 band. Just let me know!

panyingyun, 2016-07-28:
@brocaar    very thank you, please provide  me 470-510 band pre-conmpiled binaries for test, thanks.

brocaar, 2016-07-28:
here you go! https://www.dropbox.com/sh/amul3hcxz8l1s3h/AAA_u-D-F0MuuzV73VyeXABwa?dl=0

Mehradzie, 2016-07-29:
Is there a list of ISM bands defined in different countries ? :)
I am still confused about the ISM band range in China as China in the LoraWAN standard is defined for 779-787MHz ISM Band.

brocaar, 2016-07-29:
The list in the LoRaWAN specs :-) In China there are two ISM band ranges implemented:

* CN470 - 510
* CN779 - 787

For the latter one, I've created a new issue to track it's progress: https://github.com/brocaar/loraserver/issues/50.

brocaar, 2016-08-01:
@panyingyun were you able to do some testing with the binaries? Looking forward to the results :-)

Mehradzie, 2016-08-01:
"7.6  CN 470-510MHz Band" :astonished: 
How on earth I have missed this. It's totally missing from my paper copy version. *Mystery*

my bad.

panyingyun, 2016-08-04:
@brocaar  CN470 - 510  pre-conmpiled binaries   uplink data and activation-by-personalization (ABP)   test ok.    Now do not know how to test  Downlink data  and OTAA

ivajloip, 2016-08-04:
One way is to send confirmed data from your end-device. If you receive the confirmation, downlink communication works. More complicated way is through pushing data to some mqtt topic. See [this][1] for more information.

[1]: https://docs.loraserver.io/loraserver/sending-data/

brocaar, 2016-09-28:
@panyingyun looking forward to your downlink test results. Did you manage to get it working? Would be great if we can add this band to LoRa Server :-)

panyingyun, 2016-09-29:
@brocaar  You can add this band to LoRa Server now. We test all send or receive data OK(uplink /downlink /OTAA/ABP).

brocaar, 2016-09-29:
Nice! Thanks for testing :-)

closed 2016-09-30 by brocaar

brocaar, 2016-09-30:
This band will be included in the next release (0.12.2).

shouhum, 2022-05-11:
470-510 uplink is normal, downlink cannot be used, where do I need to debug

commit 22d403bf (2016-09-30): Update lorawan vendor with CN 470-510 ISM band config.

Thanks @panyingyun and @yzk0281 for testing. Closes #42.