brocaar/chirpstack-network-server #44: I want to store the uplink-data into Postgresql，where is the proper location to put my code

yzk0281, 2016-07-14:
Is it in func handleCollectedDataUpPackets() ?
Good news is that it seems the PHYPayload decrypt already done.
Anyone already do this job?

ivajloip, 2016-07-14:
Normally you would like to have a separate application that subscribes on some MQTT topic (see [this][1] for more information) and that pushes the data to the database.

If you don't want to create another application, you can add implementation of [ApplicationBackend][] (check [this][2] one for example), but then you might have problems merging when the loraserver changes in future versions...

[1]: https://docs.loraserver.io/loraserver/receiving-data/
[2]: https://github.com/brocaar/loraserver/blob/master/internal/loraserver/application/mqttpubsub/backend.go
[ApplicationBackend]: https://github.com/brocaar/loraserver/blob/master/internal/loraserver/interfaces.go#L19

yzk0281, 2016-07-15:
in func handleCollectedDataUpPackets()
how to read the content of macPL.FRMPayload？

brocaar, 2016-07-18:
Best would, as @ivajloip already mentioned, to subscribe to your node(s) MQTT topic(s) (`application/[AppEUI]/node/[DevEUI]/rx`) and persist the data yourself.

VirTERM, 2016-07-18:
I would expect peristemcy to be handled by the application server. In my case, I grab the data from Mqtt, interpret it, insert it into InfluxDB and then connect Grafana to it for visual presentation. With InfluxDB you can control retention period etc, so you could have different classes of service etc.

closed 2016-08-10 by brocaar

tusharvb19, 2018-08-27:
@VirTERM how you are connecting from mqtt to influxdb? Using python?