brocaar/chirpstack-network-server #49: Implement 433 MHz ISM Band for Rest of the World

mkurnadi, 2016-07-23:
This is a separate thread to request implementation for 433 MHz ISM Band for Rest of the World. Thanks

brocaar, 2016-07-24:
Hi @mkurnadi, I've added a test-implementation in https://github.com/brocaar/loraserver/tree/test_eu433. For convenience you can download pre-compiled binaries here: https://www.dropbox.com/sh/27n77z9i37jyzob/AACKICcESHiMZ1W5YN6akOM9a?dl=0. To start loraserver with the EU 433 band, use the `--band EU_433` flag.

Could you let me know if the following works?

- [ ] data up (node -> server)
- [ ] data down (server -> node)
- [ ] otaa (over-the air activation)

brocaar, 2016-07-29:
Hi @mkurnadi I'm looking forward to your test-results :-)

mkurnadi, 2016-08-01:
hi @brocaar, I am getting multi channel gateway and just arrived from overseas. I have been using single channel gateway in last few month. I will try the multi GW with your development software. Many thanks.I will update you again.

dashracer, 2016-08-05:
Hi @brocaar, I'm helping @mkurnadi to setup loraserver.
I've followed your documentation and install this specific build for 433 band.

I've executed default start command and the server is working and can accessed from browser.
but if I try to add --band option
sudo systemctl start loraserver --band 443_EU

this is the response

systemctl: unrecognized option '--band'

How can I solved this?

Thanks

brocaar, 2016-08-05:
You're passing the argument to systemctl, not loraserver. You should change
the loraserver.service file and add the flag there :)

dashracer, 2016-08-06:
Hi @brocaar, the service cannot start.

I've updated the service 
```
[Unit]
Description=loraserver
After=network.target

[Service]
User=loraserver
Group=loraserver
ExecStart=/usr/local/bin/loraserver --band 433_EU
Restart=on-failure

[Install]
WantedBy=multi-user.target

```

This is the log

```
Aug 06 02:00:54 dev loraserver[38877]: time="2016-08-06T02:00:54Z" level=fatal msg="lorawan/band: band 433_EU is undefined"
Aug 06 02:00:54 dev systemd[1]: loraserver.service: Main process exited, code=exited, status=1/FAILURE
Aug 06 02:00:54 dev systemd[1]: loraserver.service: Unit entered failed state.
Aug 06 02:00:54 dev systemd[1]: loraserver.service: Failed with result 'exit-code'.
Aug 06 02:00:54 dev systemd[1]: loraserver.service: Service hold-off time over, scheduling restart.
Aug 06 02:00:54 dev systemd[1]: Stopped loraserver.
Aug 06 02:00:54 dev systemd[1]: Started loraserver.
Aug 06 02:00:54 dev loraserver[38884]: time="2016-08-06T02:00:54Z" level=fatal msg="lorawan/band: band 433_EU is undefined"
Aug 06 02:00:54 dev systemd[1]: loraserver.service: Main process exited, code=exited, status=1/FAILURE
Aug 06 02:00:54 dev systemd[1]: loraserver.service: Unit entered failed state.
Aug 06 02:00:54 dev systemd[1]: loraserver.service: Failed with result 'exit-code'.
Aug 06 02:00:54 dev systemd[1]: loraserver.service: Service hold-off time over, scheduling restart.
Aug 06 02:00:54 dev systemd[1]: Stopped loraserver.
Aug 06 02:00:54 dev systemd[1]: Started loraserver.
Aug 06 02:00:54 dev loraserver[38890]: time="2016-08-06T02:00:54Z" level=fatal msg="lorawan/band: band 433_EU is undefined"
Aug 06 02:00:54 dev systemd[1]: loraserver.service: Main process exited, code=exited, status=1/FAILURE
Aug 06 02:00:54 dev systemd[1]: loraserver.service: Unit entered failed state.
Aug 06 02:00:54 dev systemd[1]: loraserver.service: Failed with result 'exit-code'.
Aug 06 02:00:55 dev systemd[1]: loraserver.service: Service hold-off time over, scheduling restart.
Aug 06 02:00:55 dev systemd[1]: Stopped loraserver.
Aug 06 02:00:55 dev systemd[1]: Started loraserver.
Aug 06 02:00:55 dev loraserver[38896]: time="2016-08-06T02:00:55Z" level=fatal msg="lorawan/band: band 433_EU is undefined"
Aug 06 02:00:55 dev systemd[1]: loraserver.service: Main process exited, code=exited, status=1/FAILURE
Aug 06 02:00:55 dev systemd[1]: loraserver.service: Unit entered failed state.
Aug 06 02:00:55 dev systemd[1]: loraserver.service: Failed with result 'exit-code'.
Aug 06 02:00:55 dev systemd[1]: loraserver.service: Service hold-off time over, scheduling restart.
Aug 06 02:00:55 dev systemd[1]: Stopped loraserver.
Aug 06 02:00:55 dev systemd[1]: Started loraserver.
Aug 06 02:00:55 dev loraserver[38903]: time="2016-08-06T02:00:55Z" level=fatal msg="lorawan/band: band 433_EU is undefined"
Aug 06 02:00:55 dev systemd[1]: loraserver.service: Main process exited, code=exited, status=1/FAILURE
Aug 06 02:00:55 dev systemd[1]: loraserver.service: Unit entered failed state.
Aug 06 02:00:55 dev systemd[1]: loraserver.service: Failed with result 'exit-code'.
Aug 06 02:00:55 dev systemd[1]: loraserver.service: Service hold-off time over, scheduling restart.
Aug 06 02:00:55 dev systemd[1]: Stopped loraserver.
Aug 06 02:00:55 dev systemd[1]: loraserver.service: Start request repeated too quickly.
Aug 06 02:00:55 dev systemd[1]: Failed to start loraserver.
```

brocaar, 2016-08-06:
Nice! Could you confirm both data up and down is working (e.g. confirmed data up or OTAA)?

mkurnadi, 2016-08-13:
Dear @brocaar, we still trying to test this. However we would like to know if lora-gateway-bridge need to be installed in the server or in the gateway ?

brocaar, 2016-08-13:
That is up to you to decide, both ways are possible. See also https://docs.loraserver.io/lora-gateway-bridge/getting-started/#strategies

k30r88, 2016-08-14:
@brocaar we have successfully do data up (node -> server). We monitor the payload by subscribe to topic gateway/b827ebffffa39301/rx 

{
  "rxInfo": {
    "mac": "b827ebffffa39301",
    "time": "0001-01-01T00:00:00Z",
    "timestamp": 3081371443,
    "frequency": 433500000,
    "channel": 0,
    "rfChain": 0,
    "crcStatus": 1,
    "codeRate": "4/5",
    "rssi": -88,
    "loRaSNR": -1,
    "size": 48,
    "dataRate": {
      "modulation": "LORA",
      "spreadFactor": 7,
      "bandwidth": 125
    }
  },
  "phyPayload": "QG4E8geABAABEJrIQ8DdJLtPAzXd2cgnKSYhjTMkQdHR/AF0oc1PEHC5q8FJ6T9O"
}
gateway id : b827ebffffa39301 in lora-server web app is known as what ?
We still trying to figure out how to retrieve data from lora-server API.

brocaar, 2016-08-14:
Hi @k30r88, LoRa Server doesn't have any knowledge about gateways. However, you should create an application and node in LoRa Server (web-interface or API). When configuring your node with the same `AppEUI`, `DevEUI` and `AppKey` you should be able to do an OTAA. Could you let me know if that works?

Receiving data: https://docs.loraserver.io/loraserver/receiving-data/
Sending data: https://docs.loraserver.io/loraserver/sending-data/

brocaar, 2016-08-14:
Hi @dashracer I'm looking forward to hear from you if you were be able to send any data from the server to the node and / or were able to perform an OTAA.

k30r88, 2016-08-14:
@brocaar . I work together with @dashracer , Application have successfully received data on the server side :  
[image]

However we test it using single channel device, frequency only at 4335000.
We will try to perform OTAA. Perhaps you can guide us some tips ?

brocaar, 2016-08-14:
@k30r88 ah great! Regarding OTAA, just provision your node with the `DevEUI`, `AppEUI` and `AppKey` and perform a `mac join otaa` (for the RN2483, this command might be different if you're using a different node). You should see a message on the `application/+/node/+/join` mqtt topic, and your node should be able to send uplink data afterwards.

shmrymbd, 2016-08-20:
hi @k30r88 , can you share where is the file you make the changes to support 433? I need to start one similar for 919-928 range. Thanks.

brocaar, 2016-08-20:
@shmrymbd see this diff for the changes needed: https://github.com/brocaar/loraserver/compare/test_eu433?expand=1

Note that the band config lives here: https://github.com/brocaar/lorawan/tree/master/band

brocaar, 2016-09-28:
@k30r88 @dashracer could you please tell us if both the uplink and downlink is working with this config? Would be great it we finally can merge this band :-)

k30r88, 2016-09-30:
@brocaar uplink seems to be working, we will try downlink next week. Do you have any suggestion on how to do downlink test ? Is downlink similar to OTAA ?

brocaar, 2016-09-30:
Downlink is sending data to your node :-) But if you managed to get OTAA
working, you already sent a downlink payload to your node (in this case not
data but a join-accept message).

ksuie, 2016-12-13:
OS: Arch Linux
loraserver: 0.8.2-1-gf6e02d3
lora-gateway-bridge: 2.1.2

RPi 2: Arch Linux ARM
gateway: [RHF0M301](http://www.risinghf.com/product/rhf0m301/?lang=en)
packet_forwarder: 2.2.1

[Moteino MEGA](https://lowpowerlab.com/shop/moteinomega) + RN2483(RN2483 1.0.1 Dec 15 2015 09:38:09)
Arduino library: TheThingsNetwork-0.17.0, pached for 433MHz

Both ABP and OTAA join ok, and uplink/downlink ok
When I use ABP and reset Moteino, must goto LoRa Server UI, set FCnt to zero

But when I use packet_forwarder-3.1.0, I got the error message:
JSON down: {"txpk":{"imme":false,"tmst":290279995,"freq":433.175,"rfch":0,"powe":10,"modu":"LORA","datr":"SF7BW125","codr":"4/5","ipol":true,"size":17,"data":"IDa54Qt8+kbrNUOSMVNrq74="}}
ERROR: Packet REJECTED, unsupported RF power for TX - 7

brocaar, 2017-01-08:
@ksuie hmm, I'm not sure if that error is an issue with the packet-forwarder config or with the band config (given that it worked before). I'll add the EU 433 band config in the next release (+ some more bands). In case you're still experiencing any errors, could you open a new issue?

closed 2017-01-08 by brocaar

gclairec, 2018-02-23:
Hi is the 433mhz support still working?