brocaar/chirpstack-network-server #52: Join accept delay

BWhitten, 2016-07-31:
I think in join_request_accept.go I see that join-accept messages are loaded with the nodes RXDelay parameter (LoRa spec says 1s) where the node should probably use a separate JoinAcceptDelay parameter (LoRa spec says 5s) for the join accept.
Motes adhering to the LoRa spec fail to pickup the join-accept message if it's sent like a standard data packet at 1s.

brocaar, 2016-07-31:
Hi @BWhitten, could you post a link to the link where you see the `RXDelay` being used for the join-accept transmission timestamp? As far as I'm aware, I'm always using `common.Band.JoinAcceptDelay1`: https://github.com/brocaar/loraserver/blob/master/internal/loraserver/join_request_accept.go#L178

The `RXDelay` is set to the `NodeSession` ([here](https://github.com/brocaar/loraserver/blob/master/internal/loraserver/join_request_accept.go#L124)) and to the `JoinAcceptPayload` ([here](https://github.com/brocaar/loraserver/blob/master/internal/loraserver/join_request_accept.go#L147)) to provision the node with this `RXDelay` setting and for LoRa Server so that when it receives data, it knows based on the `NodeSession` which `RXDelay` to use for downlink (data) responses. This setting is (as far as I'm aware of) only used after the join-accept response.

BWhitten, 2016-08-01:
Ahh I see, I was mistaken when I saw those above lines. I thought the JoinAcceptPayload was what was being sent out to the gateway. I shall close this, thanks.

closed 2016-08-01 by BWhitten