brocaar/chirpstack-network-server #54: confirmation not received

danielh93, 2016-08-01:
Hello!

First of all thank you for you efforts for building up the lora server. Everything easy to use and have a great documentation. 

I have a iC880A + Rasp Pi 3 gateway with Microchip LoRa Mote (RN2483). I using the Semtech lora packet forwarder (lora_pkt_fwd) with your gateway_bridge and server. The gateway and the endpoint are few centimeter away from each other.

I have a problem with the confirmed messages + OTAA joining (I think its the same problem). Most of the time when I send a confirmed message I can see my packet arrived successfully (CRC_OK) but after that the server do not send back a confirm message so I got a mac_err as respond from the RN2483. I got a confirmed message about every 8th attempt. I think it is the same with OTAA my packets arrived, but the server do not respond to them and I can join about every 6th attempt.

Sometimes I got 2, 3 confirmation from the server in row (very rarely).

I couldn't see any interesting log info in the server log. Seems like my packet is never arrived there. Do you have any idea whats going wrong?

ivajloip, 2016-08-01:
Hi,

I think it will be useful if you can provide

1. logs from loraserver
2. logs from lora-gateway-bridge
3. logs from packet_forwarder
4. versions of the above
5. any other relevant information

this can provide some additional information why there is no reply received by the end-node. Whether the loraserver, the gateway-bridge or the packet forwarder didn't send it or the end-node simply did not receive it due to some other problem (this happens sometimes to me, I had to use an external device to check that the message was really sent and the end-node simply did not receive it).

Mehradzie, 2016-08-01:
@danielh93 are you in the close range from your gateway? ( I mean both-on-your-desk sort of range)
Because we have some issues with the Downlink here in Australia when we get closer to the margins of the coverage but nothing in the close range.

closed 2016-09-03 by brocaar