brocaar/chirpstack-network-server #60: could not setup mqtt backend: Identifier rejected

Daniel-dk, 2016-08-11:
is there a way to set the Client ID ?
We're connecting it to VerneMQ and getting the "Identifier rejected" response to a join for both loraserver and lora-gateway-bridge

I'm not sure if our sever is not configured properly or the client connect is different from what it expects and mosquitto is ok with it

 - [MQTT-3.1.3-8]
If the Client supplies a zero-byte ClientId with CleanSession set to 0, the Server MUST respond to the >CONNECT Packet with a CONNACK return code 0x02 (Identifier rejected) and then close the Network >Connection.
 - [MQTT-3.1.3-9]
If the Server rejects the ClientId it MUST respond to the CONNECT Packet with a CONNACK return >code 0x02 (Identifier rejected) and then close the Network Connection

brocaar, 2016-08-11:
Thanks for reporting this issue. I'll look into it later today or tomorrow!

brocaar, 2016-08-11:
The clients are connecting with an empty client id since you might want to run multiple instances of LoRa Server, so I think it is easier leaving it blank and let the broker generate a random id. I've checked and `CleanSession` is `true` by default, so that is not the reason of the rejection (in reaction to your original issue). 

I haven't had the time yet to setup VerneMQ, but I can't reproduce your issue with Mosquitto. Would you mind sharing your VerneMQ config (+ the version of VerneMQ you're using) so I'm able to reproduce your setup?

brocaar, 2016-08-11:
Additionally, could you verify with for example MQTTLens that you're able to connect with your credentials (test with both an empty client id and setting a client id) and subscribe to for example `application/+/node/+/tx`?

Daniel-dk, 2016-08-12:
will do, I'll post those up later today

Daniel-dk, 2016-08-12:
All tests were performed with username and password login.

While setting the brokers "allow_anonymous" = true and "allow_multiple_session" false :

anon and !multiple_session : connect and subscribe with ID success
anon and !multiple_session : connect and subscribe without ID success
anon and !multiple_session : connect with duplicate ID failure ( switches between clients -disconnects one to allow the other)

When I set "allow_anonymous" = false we can only connect when sending a client ID,
!anon, !multiple_session : with ID success
!anon, !multiple_session : without ID fail

Setting "allow_multiple_sessions" = true doesn't seem to do anything as the server still does not allow multiple clients with the same clientID to connect simultaneously

We've decided to install a separate instance of VerneMQ for use by ONLY the loraserver and  lora-gateway-bridge ( so we can use it with anonomous logins enabled ) and then have a seperate broker with more strict security for the applications to access.

brocaar, 2016-08-12:
[image] http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.html

I think that is the source of the issue you're experiencing. Following the specs a broker doesn't have to support an empty `ClientId`, so probably it is better when I generate a random `ClientId` at the client (LoRa Server / LoRa Gateway Bridge) instead of relying on the broker.

RakshitAdmar, 2017-08-01:
I'm sorry, is the bug fixed? I'm getting the same message on my bridge logs. However, the app-server is successfully receiving data.

aquadize, 2017-08-28:
Dear Orne, hwo are you?

I am facing the same issue and it seems to be arround line 115 on loraserver/vendor/github.com/eclipse/paho.mqtt.golang/packets/connect.go where the CleanSession flag is incorrectly checked, IMHO.

Current:
	if len(c.ClientIdentifier) == 0 && c.CleanSession {
		//Bad client identifier
		return ErrRefusedIDRejected
	}

Should be:
	if len(c.ClientIdentifier) == 0 && **!c.CleanSession** {
		//Bad client identifier
		return ErrRefusedIDRejected
	}

Regards,
Ariel

commit b4627b62 (2018-03-05): Expose MQTT client id, qos and clean-session options.

Closes #60.
commit 2bc35c30 (2018-03-05): Expose MQTT client id, qos and clean-session options.

Closes #60.
commit ef5e48ab (2018-03-05): Expose MQTT client id, qos and clean-session options.

Closes #60.
commit bf21e656 (2018-03-06): Expose MQTT client id, qos and clean-session options.

Closes #60.
closed 2018-03-06 by brocaar