brocaar/chirpstack-network-server #61: Cannot create new node because of DevEUI size check

iBrick, 2016-08-11:
Hi,

After installing new version (looks like this bug appeared about 7-10 days ago) I can't create a new node. 
If I put a size of 8 byte in DevEUI it shows message 'exactly 16 bytes are expected', when I put 16 bytes in it it says 'exactly 8 bytes are expected'.
[image]
[image]

Also I suggest to exclude this check:
if ns.DevAddr.NwkID() != a.ctx.NetID.NwkID() {
		return grpc.Errorf(codes.InvalidArgument, "DevAddr must contain NwkID %s", hex.EncodeToString([]byte{a.ctx.NetID.NwkID()}))
	} 
from file /internal/api/node_session.go because it is a usual situation when you buy a node with fixed DevAddr written on it and this check doesn't allow to add this node.

brocaar, 2016-08-11:
The error you're seeing is related to the `AppKey` field, not the `DevEUI`. Please note that the web-interface is really basic (at some point, there will be a better one, but I wanted to provide a gui on top of the API at least).

Regarding your second point, I'm not sure if it is common that a node has a fixed `DevAddr`. The check is there since the `NwkID` prefix will be used for roaming. I've heard this is now under discussion and might be specified further in the upcoming LoRaWAN spec.

(one option I'm considering is to make the `NetID` optional, so that the full 32 bit address space can be used, when `NetID` and `NwkID` is not important)

iBrick, 2016-08-11:
Yes, you right, there's no problem in node creation here. Regarding DevAddr, at least in Loriot server they allow any DevAddr. And to me it seems that it is very convenient if you print on a sticker an addr of node and put it on sensor so that you may see which is sending data on gateway or server without connecting to it.

brocaar, 2016-08-11:
Wouldn't the `DevEUI` make more sense (this EUI is as well being used as identifier to the application)? Please note that after an OTAA, the device will get a new random `DevAddr`.

iBrick, 2016-08-11:
Agree, it is not a simple question. In my case we use only ABP as it is more secure and DevAddr is the only thing you see from gateway to understand who is the sender. Anyway it is not a bug and you may leave it or remove as you wish. For me it is easier if it is removed.

brocaar, 2016-08-11:
Making the `NetID` (and thus the `NwkID` prefix in the `DevAddr`) optional might be the best solution I think :-) I'll look into it. Mind if we close this issue and track the `DevAddr` issue in #56?

iBrick, 2016-08-11:
Agree =)

closed 2016-08-11 by iBrick