brocaar/chirpstack-network-server #66: Need to delete Node Session when deleting Node

jamiechard, 2016-08-18:
During the deletion of a Node (in internal/api/node.go) can you also delete the Node Session as well please. Leaving the Node Session causes data clashes later if the Node is recreated for a different App.

Maybe add the following after line 226 of internal/api/node.go

 		return nil, grpc.Errorf(codes.Unauthenticated, "authentication failed: %s", err)
 	}
 
+	ns, err := storage.GetNodeSessionByDevEUI(a.ctx.RedisPool, eui)
+	if err != nil {
+		return nil, grpc.Errorf(codes.Unknown, err.Error())
+	}
+
+	if err := storage.DeleteNodeSession(a.ctx.RedisPool, ns.DevAddr); err != nil {
+		return nil, grpc.Errorf(codes.Unknown, err.Error())
+	}
+
 	if err := storage.DeleteNode(a.ctx.DB, a.ctx.RedisPool, eui); err != nil {
 		return nil, grpc.Errorf(codes.Unknown, err.Error())
 	}

brocaar, 2016-08-19:
Good one, will implement. However, the above won't work since it is not given that a node-session exists for the given node. I'm now implementing a `DeleteNodeSessionIfExistsByDevEUI` function for this. As well it is better to hook this up to the `storage.DeleteNode` function :-)

brocaar, 2016-08-19:
Actually, this was already implemented, see: https://github.com/brocaar/loraserver/blob/master/internal/storage/node.go#L93

closed 2016-08-19 by brocaar

jamiechard, 2016-08-22:
Hi,

Thank-you for looking at this.

Can I ask if ABP is working with this change please. I dropped and then created a completely new “loraserver” DB, and flushed out Redis. However, when I try to configure a new ABP node though I get a Redis error.

Regards,

Jamie