brocaar/chirpstack-network-server #67: How to deploy lora-gateway-bridge and loraserver as a cluster

panyingyun, 2016-08-19:
1.  now for demo test, we deploy  "lora-gateway-bridge"   "loraserver"  "mosquitto" "redis server" "PostgreSQL server" "http server"  in one server or different server

brocaar, 2016-08-22:
You're right. Right now, each instance will receive the packet, but loraserver will de-duplicate the packets (as the same packet might be received by multiple gateways), so I would expect that the end-user will receive only one payload. However I realise now that the `gw_count` will be incorrect as this will be multiplied by the number of loraserver instances.

I'll look into this (please leave this issue open so I can keep track of it).

Regarding the lora-gateway-bridge, this is not an issue. Each bridge instance will only subscribe to the topics related to the gateways for which it manages a connection for. Keep in mind that when adding a loadbalancer in front of your bridge cluster, that it should persist the session of the gateway to the same backend.

brocaar, 2016-08-23:
de-duplication for 2 & 3 are already handled by the backends (thus for downlink payloads and mac commands).

panyingyun, 2016-08-24:
thanks, I found this message in github.com/brocaar/loraserver/internal/backend/application/backend.go 
and github.com/brocaar/loraserver/internal/backend/controller/backend.go 
"
        // Since with MQTT all subscribers will receive the downlink messages sent
	// by the application, the first loraserver receiving the message must lock it,
	// so that other instances can ignore the message.
	// As an unique id, the Reference field is used.
"

yzk0281, 2016-09-12:
My opinion:
Multiple loraserver can sub/pub to different "gateway/+/rx"  like we can have 16 loraservers sub/pub to "gateway/+0/rx"  "gateway/+1/rx"  "gateway/+2/rx"  ....   "gateway/+f/rx"
thus each loraserver will only handle 1/16 of the burden, maybe this is a optional way to load-balance for loraserver
Brocaar's  idea of lora-gateway-bridge is right~
Decoupling mentioned in #68 is necessary. In my vision we should have loraserver as a pure and independent data-rx service. Currently I use  a rest-api framework to realize the rest function of loraserver other than data-rx, which talk to loraserver through postgres & redis.
And more, we also need to accomplish  redis-cluster  mqtt-cluster  and  postgres-cluster, which are important to handle the huge amount of nodes.

By the way, thanks to Sir brocaar's brilliant work of loraserver, which is very very helpful and admirable.

brocaar, 2016-09-28:
The de-duplication of gateway data is fixed in 0f9541b1fe0a81bfc553531876e9f9f3a2c2bf0d, so with 0.12.1, the `gw_count` should be correct when running multiple LoRa Server instances.

closed 2016-09-28 by brocaar