brocaar/chirpstack-network-server #70: Who is responsible for downlink duty-cycle management?

brocaar, 2016-08-19:
For uplink, I think it is the responsibility of the node that it does not exceed its duty-cycle (e.g. for the EU ISM band). Which component should be responsible for downlink? Is this the network server, or is it the responsibility of the gateway (packet_forwarder)?

nestorayuso, 2016-08-19:
Is the network server responsability, but up to date none of the network servers implement it.

ADR, Transmit-Power-Control, downlink gateway election, RX1 or RX2 election, channel planning and downlink duty-cycle are highly coupled and I think must by implemented by the same module.

ivajloip, 2016-08-19:
I agree with @nestorayuso. In my mind, the gateway should return an error if the duty cycle will be exceeded (i.e. the network server is misbehaving), but I doubt that it's implemented.

closed 2016-09-03 by brocaar