brocaar/chirpstack-network-server #73: I want to know build the loraserver code- Can anyone guide me.?

mruthk, 2016-08-23:
Could you please let me know the steps which need to be followed in order to Build this loraserver code in go language as I am not much familiar with Go language.

Please help. Thanks

brocaar, 2016-08-26:
Easiest way is to use `docker-compose`, e.g. `docker-compose run --rm loraserver bash`, that will setup the whole environment (in Docker) for you.

k4jt, 2016-08-31:
Basic steps for build loraserver

Create project directories
`$ mkdir -p network-server/src/github.com/brocaar/`
`$ cd network-server/`

Setup GOPATH env variable
`$ export GOPATH=$(pwd)`
Check that GOPATH shows current directory location
`$ echo $GOPATH`

Clone project
`$ cd src/github.com/brocaar/`
`$ git clone https://github.com/brocaar/loraserver.git`

Prepare for build
`$ cd loraserver/`
Edit Makefile to change 'GOOS ?= linux' to your operarion system (linux, darwin, windows) and 'GOARCH ?= amd64' if needed
`$ vim Makefile`

Build project
`$ make build`

Check result at build directory
`$ ls -l build/`

erfantaherkhani, 2016-09-18:
@k4jt  I've already followed your instruction, but in the last part when i run "make build" it refers some error:
 $ make build
/usr/local/go/bin/go: 1: /usr/local/go/bin/go: ELF: not found
/usr/local/go/bin/go: 2: /usr/local/go/bin/go: Syntax error: word unexpected (expecting ")")
Compiling source for linux amd64
/usr/local/go/bin/go: 1: /usr/local/go/bin/go: ELF: not found
/usr/local/go/bin/go: 2: /usr/local/go/bin/go: Syntax error: word unexpected (expecting ")")
make: *** [build] Error 2

could you please guide me to solve this?

k4jt, 2016-09-18:
Hi, @erfantaherkhani 
I'm not sure what's the reason of these errors, can you tell what operation system are you using and what kind 32-bit or 64-bit? you can check it with:
$ uname -a
then check what is bit version of golang compiler:
$ cd /usr/local/go/bin
$ file go
or just
$ go version
if  Golang compiler you installed does not match your os/architecture, you can try to get another archive from https://golang.org/dl

erfantaherkhani, 2016-09-18:
@k4jt thanks for your inspiring answer, i think the problem is about go archive, i'll try it.

hecko, 2016-10-10:
Hi Everyone, the 

`make build`

yelds the following for me

`
root@loraserver:~/network-server/src/github.com/brocaar/loraserver# make build
Compiling source for linux amd64
internal/uplink/data.go:4:2: cannot find package "context" in any of:
        /root/network-server/src/github.com/brocaar/loraserver/vendor/context (vendor tree)
        /usr/lib/go-1.6/src/context (from $GOROOT)
        /root/network-server/src/context (from $GOPATH)
Makefile:8: recipe for target 'build' failed
make: *** [build] Error 1
`

Any suggestions?

k4jt, 2016-10-10:
Hi, @hecko !
Maybe problem that you use golang version 1.6, try to install 1.7

hecko, 2016-10-10:
@k4jt thats it! Thanks!

brocaar, 2016-10-12:
@k4jt and @hecko, I've added a short note to the README. @k4jt 1.6 works too, but you have to enable vendoring (see: https://blog.gopheracademy.com/advent-2015/vendor-folder/). This is enabled by default in 1.7.

wolftrax5, 2016-10-14:
will be the same steps for [lora-app-server](https://docs.loraserver.io/lora-app-server/) ?

a-farooq, 2016-11-09:
same step for lora-aap-server too.

brocaar, 2016-11-14:
I've updated the docs, hopefully this help https://github.com/brocaar/loraserver/commit/7855dcfb8736861e3e81ecd93d2d7fee1a104b14. I'll update the docs of the other projects too.

closed 2016-11-14 by brocaar

ymkins, 2016-11-16:
There is [PR99](https://github.com/brocaar/loraserver/pull/99) with Vagrantfile for quick start.

hecko, 2016-11-17:
Hete are the docker-compose sources for the full env install: 

https://github.com/hecko/brocaar-lorawan-docker

run like so:
`git clone https://github.com/hecko/brocaar-lorawan-docker.git`
`docker-compose up --build --force-recreate --abort-on-container-exit --remove-orphans`

FYI - uses the master version from git and data are not persistent

commit 7855dcfb (2016-11-14): Update build documentation.

Closes #73.