brocaar/chirpstack-network-server #74: Problems on using username/password to access the MQTT broker/server

glederer007, 2016-08-25:
I have the following setup:
1) A packet forwarder and the lora-gateway-bridge on a LORANK8 gateway,
2) The loraserver on a PC running Ubuntu 16.4,
3) An MQTT.fx client on another PC running Ubuntu 16.4.,
4) 3 pieces of Microchip RN2483 Technology Motes.
The configuration runs fine without problems, but I have a problem with the connection to the MQTT broker running under the loraserver. I have tried to set up a username/password combination that needs to be used by the MQTT client (MQTT.fx) to connect to the loraserver. Apparently the MQTT broker is not using the username/password combination (on the client side I can set it to whatever comes to my mind, it will connect anyway). The same happens at the lora-gateway-bridge side (here I forgot to set up a username/password combination, but it connected anyway). Am I missing something?

brocaar, 2016-08-26:
Seems to be an issue with you Mosquitto auth. configuration?

glederer007, 2016-08-26:
@linux4life798 has given a detailed explanation on gitter of the problem and its solution. I will try it later this weekend. I installed loraserver on a "clean and new" installation of mosquitto that is using its default settings (i.e. no passwords, etc.).

closed 2016-09-03 by brocaar