brocaar/chirpstack-network-server #75: Activation By Personalization

Romaly, 2016-08-26:
Hi guys.
Very interesting project.
On digikey I bought rn2483 starter kit and MultiConnect Conduit, and now try to connect to yours loraserver, but have problem.
As I understand this loraserver has two type of node join to network: OTAA and ABP
I've download and install loraserver for Windows (0.11.0 release), also I've install all applications which need (MQTT, SQL, etc.)
At first sight all ok, but I can not to find where switch application server to ABP join?
When I try to add node the system requred to input DevEUI, AppKey etc., but I need NodeAdd, AppSessKey and NetSessKey

PS: Sorry for my english
Best regards, 
Roman

[image]
[image]
[image]

brocaar, 2016-08-26:
This can be improved in the UI, but to ABP you have to fill in the node session fields :-)

Romaly, 2016-08-31:
Thanks, I has understood how do it!

closed 2016-09-03 by brocaar