brocaar/chirpstack-network-server #78: Choosing the right gateway on Join or Confirm

julien-ioic, 2016-09-01:
Hi,

I created a small lorawan network with 3 gateways.
Thanks a lot for this great job!!
I see a recurent problem on the server response when the message are caught by multiple gateways.

How is done the choice between the severals gateways ?
I think the better way is to select the gateway with the better SNR (the closest).

In the file join_request_accept.go.

line 190 : if err = ctx.Gateway.SendTXPacket(txPacket); err != nil {
		return fmt.Errorf("send tx packet (rx window 1) to gateway error: %s", err)
	}

Gateway is a unique object of the Context. I don't find where is referesh the context.

Thanks in advance,

Julien From Reunion Island ...

yzk0281, 2016-09-01:
"How is done the choice between the severals gateways ?
I think the better way is to select the gateway with the better SNR (the closest)."

Hi, Julien from the beautiful island, I think loraserver currently already handle this problem:
in uplink_data.go call a func named collectAndCallOnce,
and check this collectAndCallOnce in packet.go

‘ctx.Gateway’ is in loraserver/internal/loraserver/gateway/mqttpubsub/backend.go, 
it is initiated in loraserver/cmd/loraserver/main.go
just check it there~~

julien-ioic, 2016-09-01:
ok, I found it on loraserver/internal/loraserver/packet.go.

rxPackets implements the Sort Interface.
using RSSI

func (p RXPackets) Less(i, j int) bool {
	return p[i].RXInfo.RSSI > p[j].RXInfo.RSSI
}

In my log, SNR seems much accurate.

thanks.

tftelkamp, 2016-09-01:
I'd agree that SNR is more relevant than RSSI.

However, SNR easily saturates for positive values, and then you don't see differences between the gateways: they're all good.

So maybe consider an algorithm like:

 - select highest SNR
 - if SNR > 5 then select highest RSSI from all gateways with SNR > 5

The threshold (5) we could tune.

brocaar, 2016-09-03:
Thanks guys for your input! I won't fix this directly, as I'd love to get #68 out asap (it will involve a lot of code changes + changes in the api), but when it is better to use the `SNR` (combined with `RSSI`) instead of just `RSSI`, then I'm more than happy to change is :-)

brocaar, 2016-10-01:
@tftelkamp @julien-ioic so then the sorting becomes something like:

```go
func (s RXInfoSet) Less(i, j int) bool {
	// in case SNR is equal
	if s[i].LoRaSNR == s[j].LoRaSNR {
		return s[i].RSSI > s[j].RSSI
	}

	// in case the SNR > maxSNRForSort
	if s[i].LoRaSNR > maxSNRForSort && s[j].LoRaSNR > maxSNRForSort {
		return s[i].RSSI > s[j].RSSI
	}

	return s[i].LoRaSNR > s[j].LoRaSNR
}

```

Where `maxSNRForSort=5.0` for now.

brocaar, 2016-10-01:
Please see https://github.com/brocaar/loraserver/pull/88/files#diff-66307ebee4b6e63f8bc96622a5b96ccd. Looking forward to your feedback so I can merge it and put it in the next release :-)

brocaar, 2016-10-04:
This has been implemented in the 0.12.2 release. Looking forward to your feedback! Thanks again for your feedback :-)

closed 2016-10-04 by brocaar

julien-ioic, 2016-10-18:
hi,

sorry for the delay.
It seems to be perfect.
I'll check it soonly.

thanks a lot