brocaar/chirpstack-network-server #84: Dockerfile incomplete ?

siscia, 2016-09-19:
Hi,

I was re-deploying loraserver and it looks to me like the Dockerfile is incomplete.

From this commit: https://github.com/brocaar/loraserver/commit/21361d3ee5ffc6f8c925376a26e0c1d9e9335621#diff-3254677a7917c6c01f55212f86c57fbfL16
we don't add anymore the source code inside the docker container.

Am I missing something?

Cheers

Simone

brocaar, 2016-09-19:
> From this commit: 21361d3#diff-3254677a7917c6c01f55212f86c57fbfL16

Hi Simone,

Maybe I should add a comment, but the Dockerfile is not for deployments, I
use it for development and testing (running the tests).
For deployments, I would recommend to package a pre-compiled binary as a
Docker container as there is no need to ship a full Go development
environment + all the development requirements (e.g. protobuf).

siscia, 2016-09-19:
Hi :)

yeah you are completely right... However is just too easy to simply use it

No problem thought

closed 2016-09-23 by siscia