brocaar/chirpstack-network-server #85: Incorrect readings gateways counter

Romaly, 2016-09-21:
Hi

I vave only one gateway, but sometimes the server thinking that several :)
(dev_eui=1a2b3c4d07000001 gw_count=3 gw_macs="008000000000b888, 008000000000b888, 008000000000b888")

Logs:

Sep 21 17:03:31 ubuntu-server loraserver[5955]: time="2016-09-21T17:03:31+06:00" level=info msg="backend/gateway: rx packet received"
Sep 21 17:03:31 ubuntu-server loraserver[5955]: time="2016-09-21T17:03:31+06:00" level=info msg="backend/gateway: rx packet received"
Sep 21 17:03:31 ubuntu-server loraserver[5955]: time="2016-09-21T17:03:31+06:00" level=info msg="packet(s) collected" dev_eui=1a2b3c4d07000001 gw_count=2 gw_macs="008000000000b888, 008000000000b888" mtype=UnconfirmedDataUp
Sep 21 17:03:31 ubuntu-server loraserver[5955]: time="2016-09-21T17:03:31+06:00" level=info msg="backend/controller: publishing rxinfo payload" topic="application/0102030405060708/node/1a2b3c4d07000001/rxinfo"
Sep 21 17:03:31 ubuntu-server loraserver[5955]: time="2016-09-21T17:03:31+06:00" level=info msg="backend/application: publishing rx payload" topic="application/0102030405060708/node/1a2b3c4d07000001/rx"
Sep 21 17:03:31 ubuntu-server loraserver[5955]: time="2016-09-21T17:03:31+06:00" level=info msg="node-session saved" dev_addr=07000001
Sep 21 17:03:31 ubuntu-server loraserver[5955]: time="2016-09-21T17:03:31+06:00" level=info msg="backend/gateway: publishing tx packet" topic="gateway/008000000000b888/tx"
Sep 21 17:03:47 ubuntu-server loraserver[5955]: time="2016-09-21T17:03:47+06:00" level=info msg="backend/gateway: rx packet received"
Sep 21 17:03:47 ubuntu-server loraserver[5955]: time="2016-09-21T17:03:47+06:00" level=info msg="backend/gateway: rx packet received"
Sep 21 17:03:47 ubuntu-server loraserver[5955]: time="2016-09-21T17:03:47+06:00" level=info msg="backend/gateway: rx packet received"
Sep 21 17:03:47 ubuntu-server loraserver[5955]: time="2016-09-21T17:03:47+06:00" level=info msg="packet(s) collected" dev_eui=1a2b3c4d07000001 gw_count=3 gw_macs="008000000000b888, 008000000000b888, 008000000000b888" mtype=UnconfirmedDataUp
Sep 21 17:03:47 ubuntu-server loraserver[5955]: time="2016-09-21T17:03:47+06:00" level=info msg="backend/controller: publishing rxinfo payload" topic="application/0102030405060708/node/1a2b3c4d07000001/rxinfo"
Sep 21 17:03:47 ubuntu-server loraserver[5955]: time="2016-09-21T17:03:47+06:00" level=info msg="backend/application: publishing rx payload" topic="application/0102030405060708/node/1a2b3c4d07000001/rx"
Sep 21 17:03:47 ubuntu-server loraserver[5955]: time="2016-09-21T17:03:47+06:00" level=info msg="node-session saved" dev_addr=07000001
Sep 21 17:03:47 ubuntu-server loraserver[5955]: time="2016-09-21T17:03:47+06:00" level=info msg="backend/gateway: publishing tx packet" topic="gateway/008000000000b888/tx"
Sep 21 17:06:05 ubuntu-server loraserver[5955]: time="2016-09-21T17:06:05+06:00" level=info msg="backend/gateway: rx packet received"
Sep 21 17:06:05 ubuntu-server loraserver[5955]: time="2016-09-21T17:06:05+06:00" level=info msg="backend/gateway: rx packet received"
Sep 21 17:06:05 ubuntu-server loraserver[5955]: time="2016-09-21T17:06:05+06:00" level=info msg="packet(s) collected" dev_eui=1a2b3c4d07000001 gw_count=2 gw_macs="008000000000b888, 008000000000b888" mtype=UnconfirmedDataUp
Sep 21 17:06:05 ubuntu-server loraserver[5955]: time="2016-09-21T17:06:05+06:00" level=info msg="backend/controller: publishing rxinfo payload" topic="application/0102030405060708/node/1a2b3c4d07000001/rxinfo"
Sep 21 17:06:05 ubuntu-server loraserver[5955]: time="2016-09-21T17:06:05+06:00" level=info msg="backend/application: publishing rx payload" topic="application/0102030405060708/node/1a2b3c4d07000001/rx"
Sep 21 17:06:05 ubuntu-server loraserver[5955]: time="2016-09-21T17:06:05+06:00" level=info msg="node-session saved" dev_addr=07000001
Sep 21 17:06:05 ubuntu-server loraserver[5955]: time="2016-09-21T17:06:05+06:00" level=info msg="backend/gateway: publishing tx packet" topic="gateway/008000000000b888/tx"
Sep 21 17:07:04 ubuntu-server loraserver[5955]: time="2016-09-21T17:07:04+06:00" level=info msg="backend/gateway: rx packet received"
Sep 21 17:07:04 ubuntu-server loraserver[5955]: time="2016-09-21T17:07:04+06:00" level=info msg="backend/gateway: rx packet received"
Sep 21 17:07:04 ubuntu-server loraserver[5955]: time="2016-09-21T17:07:04+06:00" level=info msg="packet(s) collected" dev_eui=1a2b3c4d07000001 gw_count=2 gw_macs="008000000000b888, 008000000000b888" mtype=UnconfirmedDataUp
Sep 21 17:07:04 ubuntu-server loraserver[5955]: time="2016-09-21T17:07:04+06:00" level=info msg="backend/controller: publishing rxinfo payload" topic="application/0102030405060708/node/1a2b3c4d07000001/rxinfo"
Sep 21 17:07:04 ubuntu-server loraserver[5955]: time="2016-09-21T17:07:04+06:00" level=info msg="backend/application: publishing rx payload" topic="application/0102030405060708/node/1a2b3c4d07000001/rx"
Sep 21 17:07:04 ubuntu-server loraserver[5955]: time="2016-09-21T17:07:04+06:00" level=info msg="node-session saved" dev_addr=07000001
Sep 21 17:07:04 ubuntu-server loraserver[5955]: time="2016-09-21T17:07:04+06:00" level=info msg="backend/gateway: publishing tx packet" topic="gateway/008000000000b888/tx"
Sep 21 17:08:26 ubuntu-server loraserver[5955]: time="2016-09-21T17:08:26+06:00" level=info msg="backend/gateway: rx packet received"
Sep 21 17:08:26 ubuntu-server loraserver[5955]: time="2016-09-21T17:08:26+06:00" level=warning msg="invalid FCnt" dev_addr=07000002 dev_eui=1a2b3c4d07000002 packet_fcnt=0 server_fcnt=14
Sep 21 17:08:26 ubuntu-server loraserver[5955]: time="2016-09-21T17:08:26+06:00" level=error msg="processing rx packet error: invalid FCnt or too many dropped frames" data_base64="QAIAAAcAAAAKgHN0M5G3nu5E2E5JzV+pXvslpOKAwHC3RyJUDE6WUT8rKDGPxKM6pBtEhfzU6tDxf1wjqK/60w=="
Sep 21 17:09:51 ubuntu-server loraserver[5955]: time="2016-09-21T17:09:51+06:00" level=info msg="backend/gateway: rx packet received"
Sep 21 17:09:51 ubuntu-server loraserver[5955]: time="2016-09-21T17:09:51+06:00" level=info msg="packet(s) collected" dev_eui=1a2b3c4d07000001 gw_count=1 gw_macs=008000000000b888 mtype=UnconfirmedDataUp
Sep 21 17:09:51 ubuntu-server loraserver[5955]: time="2016-09-21T17:09:51+06:00" level=info msg="backend/controller: publishing rxinfo payload" topic="application/0102030405060708/node/1a2b3c4d07000001/rxinfo"
Sep 21 17:09:51 ubuntu-server loraserver[5955]: time="2016-09-21T17:09:51+06:00" level=info msg="backend/application: publishing rx payload" topic="application/0102030405060708/node/1a2b3c4d07000001/rx"
Sep 21 17:09:51 ubuntu-server loraserver[5955]: time="2016-09-21T17:09:51+06:00" level=info msg="node-session saved" dev_addr=07000001
Sep 21 17:09:51 ubuntu-server loraserver[5955]: time="2016-09-21T17:09:51+06:00" level=info msg="backend/gateway: publishing tx packet" topic="gateway/008000000000b888/tx"
Sep 21 17:10:46 ubuntu-server loraserver[5955]: time="2016-09-21T17:10:46+06:00" level=info msg="backend/gateway: rx packet received"
Sep 21 17:10:46 ubuntu-server loraserver[5955]: time="2016-09-21T17:10:46+06:00" level=info msg="packet(s) collected" dev_eui=1a2b3c4d07000001 gw_count=1 gw_macs=008000000000b888 mtype=UnconfirmedDataUp
Sep 21 17:10:46 ubuntu-server loraserver[5955]: time="2016-09-21T17:10:46+06:00" level=info msg="backend/controller: publishing rxinfo payload" topic="application/0102030405060708/node/1a2b3c4d07000001/rxinfo"
Sep 21 17:10:46 ubuntu-server loraserver[5955]: time="2016-09-21T17:10:46+06:00" level=info msg="backend/application: publishing rx payload" topic="application/0102030405060708/node/1a2b3c4d07000001/rx"
Sep 21 17:10:46 ubuntu-server loraserver[5955]: time="2016-09-21T17:10:46+06:00" level=info msg="node-session saved" dev_addr=07000001
Sep 21 17:10:46 ubuntu-server loraserver[5955]: time="2016-09-21T17:10:46+06:00" level=info msg="backend/gateway: publishing tx packet" topic="gateway/008000000000b888/tx"
Sep 21 17:11:03 ubuntu-server loraserver[5955]: time="2016-09-21T17:11:03+06:00" level=info msg="backend/gateway: rx packet received"
Sep 21 17:11:03 ubuntu-server loraserver[5955]: time="2016-09-21T17:11:03+06:00" level=info msg="backend/gateway: rx packet received"
Sep 21 17:11:03 ubuntu-server loraserver[5955]: time="2016-09-21T17:11:03+06:00" level=info msg="backend/gateway: rx packet received"
Sep 21 17:11:03 ubuntu-server loraserver[5955]: time="2016-09-21T17:11:03+06:00" level=info msg="packet(s) collected" dev_eui=1a2b3c4d07000001 gw_count=3 gw_macs="008000000000b888, 008000000000b888, 00800000000 [... cut ...]
Sep 21 17:11:03 ubuntu-server loraserver[5955]: time="2016-09-21T17:11:03+06:00" level=info msg="backend/application: publishing rx payload" topic="application/0102030405060708/node/1a2b3c4d07000001/rx"
Sep 21 17:11:03 ubuntu-server loraserver[5955]: time="2016-09-21T17:11:03+06:00" level=info msg="node-session saved" dev_addr=07000001
Sep 21 17:11:03 ubuntu-server loraserver[5955]: time="2016-09-21T17:11:03+06:00" level=info msg="backend/gateway: publishing tx packet" topic="gateway/008000000000b888/tx"

brocaar, 2016-09-24:
Are you running multiple instances of LoRa Server?

brocaar, 2016-09-25:
In case you're running multiple LoRa Server instances, each of them is adding the payload to the buffer which might result in what you describe. Will look into it.

commit 0f9541b1 (2016-09-28): Fix multiple LoRa Server instances processing same uplink data.

Closes #85.
closed 2016-09-28 by brocaar