brocaar/chirpstack-network-server #86: Connection issue

drhexium, 2016-09-26:
I am trying to run ver 0.12.0 but I am getting the following error:

Sep 26 13:49:36 saide-gtw loraserver[1437]: time="2016-09-26T13:49:36+03:00" level=info msg="grpc: addrConn.resetTransport failed to create client transport: connection error: desc = \"transport: dial tcp 127.0.0.1:8001: getsockopt: connection refused\"; Reconnecting to {\"127.0.0.1:8001\" <nil>}"

For information, I run this on a Raspberry Pi 3. 

BTW: version 0.11.0 seems to work fine.

Thanks!

brocaar, 2016-09-26:
That is because you need to start the lora-app-server too (which is running
by default on port 8001). Please see the changelog for instructions to
migrate from 0.11 to 0.12 as it describes how you can migrate your data and
node-sessions.

drhexium, 2016-09-26:
Brocaar,
Thanks for the immediate response. I have now started lora-app-server and I am getting the following error:
Sep 26 14:20:06 saide-gtw lora-app-server[1750]: time="2016-09-26T14:20:06+03:00" level=info msg="grpc: addrConn.resetTransport failed to create client transport: connection error: desc = \"transport: dial tcp 127.0.0.1:8000: getsockopt: connection refused\"; Reconnecting to {\"127.0.0.1:8000\" <nil>}"

loraserver (0.12.0) seems to be working fine though

Is should mention that on the same host (Raspberry Pi) I am also running the packet-forwarder and lora-gateway-bridge (in case this causes any problems).

Sorry if I am doing something stupid :-)

brocaar, 2016-09-26:
That is the other way around, the lora-app-server trying to connect to
loraserver. Please note that you can't avoid these errors as it is kind of
a chicken egg problem. However it will try to re-connect and these error
should stop as soon as both services are connected to each other.

drhexium, 2016-09-26:
Actually, I do not see these errors ever stopping even after both services have started.

Some of the errors that I am getting:

Sep 26 14:46:56 saide-gtw lora-app-server[2749]: time="2016-09-26T14:46:56+03:00" level=info msg="grpc: addrConn.resetTransport failed to create client transport: connection error: desc = \"transport: dial tcp [::1]:8080: getsockopt: connection refused\"; Reconnecting to {\"localhost:8080\" <nil>}"

Also when trying to access the configuration page through a browser pointing at serveraddress:8080 I get this:
Sep 26 14:47:14 saide-gtw lora-app-server[2749]: 2016/09/26 14:47:14 http: TLS handshake error from 10.22.71.146:55280: tls: first record does not look like a TLS handshake

Thanks

brocaar, 2016-09-26:
Ignore the 8080 error (this is an internal connection error which only
happens at startup and is automatically resolved, basically the REST api
can't connect internally to the gRPC server). You're almost there, try to
point your browser to https://serveraddress:8080/. (note the https).

drhexium, 2016-09-26:
It worked! Thanks!!!!

Just one (hopefully) final question: - How can I check the incoming data? I cannot find the "Transmitting/Receiving Data" sections on the updated documentation. Could you point to the right direction?

Thanks again!

brocaar, 2016-09-26:
Great! See https://docs.loraserver.io/lora-app-server/mqtt-topics/

drhexium, 2016-09-26:
Excellent! Keep up the good work! Thanks

closed 2016-09-26 by drhexium