brocaar/chirpstack-network-server #87: LoRaServer would send 'UnconfirmedDataDown'  when a 'UnconfirmedDataUp' was received.

john-bigz, 2016-09-29:
I wonder if it would send 'UnconfirmedDataDown' when LoRaServer received a 'UnconfirmedDataUp'.
looking at the code below: ( [loraserver/internal/downlink/data.go)](https://github.com/brocaar/loraserver/blob/master/internal/downlink/data.go#L200)
```
// uplink was unconfirmed and no downlink data in queue and no mac commands to send
if txPayload == nil && rxPacket.PHYPayload.MHDR.MType == lorawan.UnconfirmedDataUp && len(macCommands) == 0 {
    return nil
}
```
the modified code as below:
```
// uplink was unconfirmed OR (no downlink data in queue and no mac commands to send)
if rxPacket.PHYPayload.MHDR.MType == lorawan.UnconfirmedDataUp || (txPayload == nil && len(macCommands) == 0) {
	    return nil
}```

brocaar, 2016-09-29:
I think the current code is correct?

The reason is that an `UnconfirmedDataUp` doesn't need an `ACK`, thus when there is nothing to send back (`txPayload=nil` **and** `len(macCommands)=0`), we return. So all conditions must match.

When an `ConfirmedDataUp` is received, LoRa Server always needs to respond since it needs to `ACK` the received frame.

ivajloip, 2016-09-29:
I also think that the current implementation is correct. Even if the message is unconfirmed, we might still want to reply if there is a pending dataDown packet (which is not possible with the suggested modification).

john-bigz, 2016-09-29:
Yes, I made a mistake, the server still need to send pending data-down packets when it receives data-up packet, even if the data-up packet is an unconfirmedDataUp packet.

brocaar, 2016-09-29:
No worries :-) Thanks for reviewing the code! If you find any other issues
or have any doubts or questions, please let me know.

closed 2016-09-30 by john-bigz