brocaar/chirpstack-network-server #89: Strip binaries for release

siscia, 2016-10-09:
It seems to me (I could be wrong) that the binaries are not stripped when released.

Stripping comes with some tradeoff, but it is usually well worthed in the release.

The loraserver goes from 16MB to 9.2MB when stripped -> +40% reduction
Similarly the lora-app-server.

To strip the binaries is sufficient to add "-w -s" to the -ldflags.

brocaar, 2016-11-19:
Hi @siscia I'll look into it to see if there is any negative impact. For now this hasn't been an issue as usually a few more megabytes isn't an issue, but if there is no impact, then I don't see a reason why we shouldn't strip the binaries :-)

siscia, 2016-11-21:
:+1: I definitely agree that it is not a pressing issues.

brocaar, 2018-01-26:
I think it is time to actually implement this :-) I never really took the time to dive into this, but after doing some research, I don't think that normal users should see any difference when compiling with `-w -s`. I think it only matters when using external tools for debugging, but in that case one could also compile a custom binary if needed.

https://stackoverflow.com/questions/22267189/what-does-the-w-flag-mean-when-passed-in-via-the-ldflags-option-to-the-go-comman
> None of these - not -ldflags -w, not -ldflags -s, not strip -s - should affect the execution of the actual program. They only affect whether you can debug or analyze the program with other tools.

commit 10795c7f (2018-01-28): Improve build size.

Closes #89.
closed 2018-01-28 by brocaar