brocaar/chirpstack-network-server #91: Server->Sensor send with ACK forever

iBrick, 2016-10-13:
I've found that when I send downlink message from server to sensor with ACK request this request never comes to the server and thus server never stops to try and send it again and again. 
I'm not yet sure about does sensor really send ACK or not, but with Microchip RN2483 module (which has certified LoRaWAN stack on it, I have it on LoRaMote device with display and buttons) I receive downlink messages from server and can see them on LoRaMote screen but if I send it with ACK request server looks like doesn't receive ACK (at least it still keep message and queue and keep sending it forever). 

I suggest 
1) to check where there is a bug here: does server really not receiving ACK or just don't work with it properly
2) to make such messages it queue limited by number of tries. Like if it was send 10 times without ACK - give up and send next one

brocaar, 2016-10-13:
I'm unable to reproduce the issue with these steps:

1. I created a downlink payload
2. I sent an uplink payload and received the downlink payload
3. The downlink payload is now marked as `pending=true` in the db
4. After sending a second uplink payload, the downlink payload is removed from the queue

app server logs:

```
INFO[0431] downlink queue item enqueued                  dev_eui=0202020202020202 id=23    // item added to the queue
INFO[0495] handler/mqtt: publishing data-up payload      topic=application/0202020202020202/node/0202020202020202/rx
INFO[0495] downlink queue item updated                   id=23    // mark queue item as pending
INFO[0495] data-down item requested by network-server    confirmed=true dev_eui=0202020202020202 fcnt=3 id=23    // send queue item to network-server
INFO[0505] handler/mqtt: publishing data-up payload      topic=application/0202020202020202/node/0202020202020202/rx
INFO[0505] downlink queue item deleted                   id=23    // ack received
INFO[0505] downlink queue item acknowledged              dev_eui=0202020202020202
INFO[0505] handler/mqtt: publishing ack notification     topic=application/0202020202020202/node/0202020202020202/ack    // publish ack notification over mqtt
INFO[0505] data-down item requested by network-server, but queue is empty  dev_eui=0202020202020202    // the queue is now empty
```

MQTT:

```
application/0202020202020202/node/0202020202020202/rx {"devEUI":"0202020202020202","time":"0001-01-01T00:00:00Z","fPort":1,"gwCount":1,"rssi":-59,"data":"qrvM"}
application/0202020202020202/node/0202020202020202/rx {"devEUI":"0202020202020202","time":"0001-01-01T00:00:00Z","fPort":1,"gwCount":1,"rssi":-60,"data":"qrvM"}
application/0202020202020202/node/0202020202020202/ack {"reference":"abc","devEUI":"0202020202020202"}
```

iBrick, 2016-10-13:
I rechecked again. You're right, it works fine. My fault, sorry.

From logs of Lora-app-server:
1) after 1st message for which downlink is performed

Oct 13 17:10:43 innolabsGw lora-app-server[630]: time="2016-10-13T17:10:43-03:00" level=info msg="handler/mqtt: publishing data-up payload" topic="application/0000000000011111/node/0139afcf0139afcf
/rx"
Oct 13 17:10:44 innolabsGw lora-app-server[630]: time="2016-10-13T17:10:44-03:00" level=info msg="downlink queue item updated" id=99
Oct 13 17:10:44 innolabsGw lora-app-server[630]: time="2016-10-13T17:10:44-03:00" level=info msg="data-down item requested by network-server" confirmed=true dev_eui=0139afcf0139afcf fcnt=33 id=99

2) After 2nd message it is deleted from queue:

Oct 13 17:11:19 innolabsGw lora-app-server[630]: time="2016-10-13T17:11:19-03:00" level=info msg="handler/mqtt: publishing data-up payload" topic="application/0000000000011111/node/0139afcf0139afcf
/rx"
Oct 13 17:11:20 innolabsGw lora-app-server[630]: time="2016-10-13T17:11:20-03:00" level=info msg="downlink queue item deleted" id=99
Oct 13 17:11:20 innolabsGw lora-app-server[630]: time="2016-10-13T17:11:20-03:00" level=info msg="downlink queue item acknowledged" dev_eui=0139afcf0139afcf

3) After 3rd message queue is empty
Oct 13 17:11:20 innolabsGw lora-app-server[630]: time="2016-10-13T17:11:20-03:00" level=info msg="handler/mqtt: publishing ack notification" topic="application/0000000000011111/node/0139afcf0139afc
f/ack"
Oct 13 17:11:20 innolabsGw lora-app-server[630]: time="2016-10-13T17:11:20-03:00" level=info msg="data-down item requested by network-server, but queue is empty" dev_eui=0139afcf0139afcf

Sorry for false alarm.

closed 2016-10-13 by iBrick