brocaar/chirpstack-network-server #92: LoRaWAN server minimum footprint

fcolumbu, 2016-10-25:
How small a footprint can the brocaar LoRaWAN server fit in with all dependent components included?  

I see that your server uses 3.62 MB in the case of  loraserver_0.12.3_linux_amd64.tar.gz but then I would also need  Mosquito and Redis and its dependencies but would like to understand what to expect  with the total footprint. How much RAM should I plan for as well?

Has anyone created a Docker image? 

If so, how large is it?

Anyone using Alpine to implement this  LoRaWAN server in a very small size?

ivajloip, 2016-10-25:
Hi,

We ([Acklio](http://www.ackl.io/)) are building a solution for LoRaWAN based on this project. We have tested our software on ARM and x86 equipment and it seems to need less than 128 MB RAM and the size of the executables is around 35 MB. If you have Docker anyway on your system, the size can be slightly reduced. To achieve this we use linux Alpine and musl, so we are sure our modified version works well under it.

If you want to learn more, please do not hesitate to contact us.

ivajloip, 2016-10-25:
BTW in connection to this, you might want to check our docker hub repositories [here](https://hub.docker.com/u/acklio/dashboard/). If you want, I think I will be able to give you the *Dockerfiles* for the images there.

brocaar, 2016-11-13:
@fcolumbu 

I’m not able to directly answer your question, but to give you a few hints what might increase the memory footprint (I haven’t done any benchmarking on this yet):

LoRa Server:
Stores received packets in Redis for de-duplication (short expiration)
Stores node-sessions in Redis (long expiration)
Stores enqueued mac-commands in Redis (long expiration)
Each received packet is dispatched by a separate Go routine (thus processing many concurrent packets might increase the footprint temporarily).

LoRa App Server:
Stores downlink packets (received over MQTT) in Redis for de-duplication (short expiration)
Stores node data and downlink queue in PostgreSQL (enqueued items are removed after transmission)

LoRa Gateway Bridge:
Stores an in-memory list of connected gateways

hecko, 2016-11-20:
To add to @brocaar 's response - I have mande docker-compose images for the complete testing environment here: 

https://github.com/hecko/brocaar-lorawan-docker

We are running the installation on the cheapest Linode virtual server without any performance bottlenecks so far.

closed 2017-01-27 by brocaar