brocaar/chirpstack-network-server #93: Geolocation solution adds GPS-free tracking capability to LoRaWAN™ networks.

panyingyun, 2016-10-27:
Is anyone Who use TODA. Try to calc geolocation by three or more  uplink data provide by gateway. 

Some refference videos and web links here:
https://www.youtube.com/watch?v=l1GmBP2zMhU
https://github.com/StevenJL/tdoa_localization/blob/master/TDOAShell.m
https://en.wikipedia.org/wiki/Multilateration

nestorayuso, 2016-10-27:
To use TDOA you need high precition timestamping
A 1st generation LoRa gateway with GPS and PPS output doen not have enough precission.
Semtech designed a 2nd generation gateway with FPGA to calculate better time stamps. The problem is the timestamping metadata is encrypted inside the gateway and you must pay a fee to semtech to desencrypt it. Semtech offer a free simple geolocation solver based on TDOA, or you can use your own or a third party one like CSEM. 3rd party ones are better because also use 3D floor planning maps to improve results and mitigate multipaths.

read:
https://www.thethingsnetwork.org/forum/t/location-by-triangulation/435
http://www.link-labs.com/lora-localization/

radekvozak, 2017-01-31:
Hi, maybe this help - but it is in Czech (1th place in IOT summer JAM 2016 in CZ).  Code is in english.

https://dl.dropboxusercontent.com/u/15330582/Summer%20Jam/JanStejskal/prezentace.pdf
https://dl.dropboxusercontent.com/u/15330582/Summer%20Jam/JanStejskal/popis-projektu.pdf
https://github.com/janstejskal/lorawan_gps_tracker-prototype
http://lorawan.zobrazit.info

closed 2017-04-25 by brocaar

boundaries-io, 2018-08-31:
https://www.youtube.com/watch?time_continue=167&v=k5u8KJHN2P0

brocaar, 2018-09-17:
Please see https://forum.loraserver.io/t/release-lora-app-server-v2-2-0-lora-geo-server-v2-0-0/2246.