brocaar/chirpstack-network-server #95: Possible to add the frame count and SNR to the output json meta data?

andrepferreira, 2016-11-02:
Is it possible to add the frame count and SNR value to the devEUI structure? It is useful to monitor the state of some of the sensors (restarts etc) and it would be useful to have it in the incoming json. But, I'm not sure if this is in line with the standard.

Something like this:

{"devEUI":"0000000079250001","time":"2016-10-16T20:10:08.202462Z","fPort":1,"gwCount":1,"rssi":-35, **"snr":9.2**, **"fcnt":236**,"data":"AYD+AAEJ9QFQACMADAAAAGZmzEHRqwfCOjqTQQAAlkKamZk/"}

glederer007, 2016-11-11:
Hello andrepferreira and TimelessNL,

I had a similar comment, but I needed only the fcntup value. To have the fcntup value in the JSON structure needs a very simple modification; you have to modify the mqtt_handler.go and the application_server.go files in **_lora-app-server_**. I attached two screenshots about where to put the modification (not the modification itself!).

1) The mqtt_handler. go contains the definition of the DataUpPayload struct. Here you need to add FCntUp as a new field (around line 25).
[image]

2)  The application_server.go actually uses the struct defined in mqtt_handler.go, here you need to add a line to give the value to FCntUp (around line 225).
[image]

Adding LoRaSNR in a similar way is not possible, you should investigate the code to find the availability of this data.

I hope my solution gives you some help.

Regards,

Gabor

brocaar, 2016-11-19:
I'll look into this asap. I agree this is helpful information to have in the JSON data.

hecko, 2016-11-19:
+1
Although I would love to see the rssi information also tied with the gateway mac to get some sense of the rssi information (and possibly snr in the future).

brocaar, 2016-11-25:
Would this work? I'm planning to release this as part of LoRa Server 0.1.4.

```json
{
    "devEUI": "0202020202020202",  // device EUI
    "rxInfo": [
        {
            "mac": "0303030303030303",                 // MAC of the receiving gateway
            "time": "2016-11-25T16:24:37.295915988Z",  // time when the package was received (GPS time of gateway, only set when available)
            "rssi": -57,                               // signal strength (dBm)
            "loRaSNR": 10                              // signal to noise ratio
        }
    ],
    "fCnt": 10,                    // frame-counter
    "fPort": 5,                    // FPort
    "data": "..."                  // base64 encoded payload (decrypted)
}
```

glederer007, 2016-11-25:
It would be perfect!

Xexell, 2016-11-25:
I'm waiting for this update. Thanks man, it would be very usefull!!

brocaar, 2016-11-25:
I've just merged the changes and tagged a new release. It will be available as a download within a few minutes (the build server is running the tests + building the packages). Keep an eye on: https://github.com/brocaar/lora-app-server/releases.

closed 2016-11-25 by brocaar