brocaar/chirpstack-network-server #96: LoRa App Server does not start.

vspherei, 2016-11-11:
The log file is as follow:

> -- Logs begin at Fri 2016-11-11 18:34:53 CET. --
Nov 11 18:36:51 AppServer systemd[1]: lora-app-server.service: Unit entered failed state.
Nov 11 18:36:51 AppServer systemd[1]: lora-app-server.service: Failed with result 'exit-code'.
Nov 11 18:36:51 AppServer systemd[1]: lora-app-server.service: Service hold-off time over, scheduling restart.
Nov 11 18:36:51 AppServer systemd[1]: Stopped lora-app-server.
Nov 11 18:36:51 AppServer systemd[1]: Started lora-app-server.
Nov 11 18:36:51 AppServer systemd[2452]: lora-app-server.service: Failed at step USER spawning /opt/lora-app-server/bin/lora-app-server: No such process
Nov 11 18:36:51 AppServer systemd[1]: lora-app-server.service: Main process exited, code=exited, status=217/USER
Nov 11 18:36:51 AppServer systemd[1]: lora-app-server.service: Unit entered failed state.
Nov 11 18:36:51 AppServer systemd[1]: lora-app-server.service: Failed with result 'exit-code'.
Nov 11 18:36:51 AppServer systemd[1]: lora-app-server.service: Service hold-off time over, scheduling restart.
Nov 11 18:36:51 AppServer systemd[1]: Stopped lora-app-server.
Nov 11 18:36:51 AppServer systemd[1]: lora-app-server.service: Start request repeated too quickly.
Nov 11 18:36:51 AppServer systemd[1]: Failed to start lora-app-server.
Nov 11 18:36:57 AppServer systemd[1]: Started lora-app-server.
Nov 11 18:36:57 AppServer systemd[2457]: lora-app-server.service: Failed at step USER spawning /opt/lora-app-server/bin/lora-app-server: No such process
Nov 11 18:36:57 AppServer systemd[1]: lora-app-server.service: Main process exited, code=exited, status=217/USER
Nov 11 18:36:57 AppServer systemd[1]: lora-app-server.service: Unit entered failed state.
Nov 11 18:36:57 AppServer systemd[1]: lora-app-server.service: Failed with result 'exit-code'.
Nov 11 18:36:58 AppServer systemd[1]: lora-app-server.service: Service hold-off time over, scheduling restart.
Nov 11 18:36:58 AppServer systemd[1]: Stopped lora-app-server.
Nov 11 18:36:58 AppServer systemd[1]: Started lora-app-server.
Nov 11 18:36:58 AppServer systemd[2459]: lora-app-server.service: Failed at step USER spawning /opt/lora-app-server/bin/lora-app-server: No such process
Nov 11 18:36:58 AppServer systemd[1]: lora-app-server.service: Main process exited, code=exited, status=217/USER
Nov 11 18:36:58 AppServer systemd[1]: lora-app-server.service: Unit entered failed state.
Nov 11 18:36:58 AppServer systemd[1]: lora-app-server.service: Failed with result 'exit-code'.
Nov 11 18:36:58 AppServer systemd[1]: lora-app-server.service: Service hold-off time over, scheduling restart.
Nov 11 18:36:58 AppServer systemd[1]: Stopped lora-app-server.
Nov 11 18:36:58 AppServer systemd[1]: Started lora-app-server.
Nov 11 18:36:58 AppServer systemd[2461]: lora-app-server.service: Failed at step USER spawning /opt/lora-app-server/bin/lora-app-server: No such process
Nov 11 18:36:58 AppServer systemd[1]: lora-app-server.service: Main process exited, code=exited, status=217/USER
Nov 11 18:36:58 AppServer systemd[1]: lora-app-server.service: Unit entered failed state.
Nov 11 18:36:58 AppServer systemd[1]: lora-app-server.service: Failed with result 'exit-code'.
Nov 11 18:36:58 AppServer systemd[1]: lora-app-server.service: Service hold-off time over, scheduling restart.
Nov 11 18:36:58 AppServer systemd[1]: Stopped lora-app-server.
Nov 11 18:36:58 AppServer systemd[1]: Started lora-app-server.
Nov 11 18:36:58 AppServer systemd[2463]: lora-app-server.service: Failed at step USER spawning /opt/lora-app-server/bin/lora-app-server: No such process
Nov 11 18:36:58 AppServer systemd[1]: lora-app-server.service: Main process exited, code=exited, status=217/USER
Nov 11 18:36:58 AppServer systemd[1]: lora-app-server.service: Unit entered failed state.
Nov 11 18:36:58 AppServer systemd[1]: lora-app-server.service: Failed with result 'exit-code'.
Nov 11 18:36:59 AppServer systemd[1]: lora-app-server.service: Service hold-off time over, scheduling restart.
Nov 11 18:36:59 AppServer systemd[1]: Stopped lora-app-server.
Nov 11 18:36:59 AppServer systemd[1]: Started lora-app-server.
Nov 11 18:36:59 AppServer systemd[2465]: lora-app-server.service: Failed at step USER spawning /opt/lora-app-server/bin/lora-app-server: No such process
Nov 11 18:36:59 AppServer systemd[1]: lora-app-server.service: Main process exited, code=exited, status=217/USER
Nov 11 18:36:59 AppServer systemd[1]: lora-app-server.service: Unit entered failed state.
Nov 11 18:36:59 AppServer systemd[1]: lora-app-server.service: Failed with result 'exit-code'.
Nov 11 18:36:59 AppServer systemd[1]: lora-app-server.service: Service hold-off time over, scheduling restart.
Nov 11 18:36:59 AppServer systemd[1]: Stopped lora-app-server.
Nov 11 18:36:59 AppServer systemd[1]: lora-app-server.service: Start request repeated too quickly.
Nov 11 18:36:59 AppServer systemd[1]: Failed to start lora-app-server.

>

glederer007, 2016-11-11:
Hello vspherei,

This happened to me when I accidentally downloaded the wrong version of the lora-app-server (linux_386 version instead of linux-arm). Sorry for the question, but are you sure you are using the correct binary version?

vspherei, 2016-11-11:
Yes, you are right. I followed the steps again and it works right now.

closed 2016-11-11 by vspherei