brocaar/chirpstack-network-server #97: getsockopt: connection refused"; Reconnecting to {"127.0.0.1:8001" <nil>}

lancezhangsf, 2016-11-15:
hi, brocaar,
i run the loraserver, but some error happen,as f
root@zhangsf-virtual-machine:/opt/loraserver/bin# ./loraserver  --net-id 010203  --band CN_470_510  
INFO[0000] starting LoRa Server                          band=CN_470_510 docs=https://docs.loraserver.io/ net_id=010203 version=0.12.3
INFO[0000] setup redis connection pool                   url=redis://localhost:6379
INFO[0000] backend/gateway: connecting to mqtt broker    server=tcp://localhost:1883
INFO[0000] connecting to application-server              ca-cert= server=127.0.0.1:8001 tls-cert= tls-key=
INFO[0000] no network-controller configured             
INFO[0000] starting api server                           bind=0.0.0.0:8000 ca-cert= tls-cert= tls-key=
INFO[0000] backend/gateway: connected to mqtt server    
INFO[0000] backend/gateway: subscribing to rx topic      topic=gateway/+/rx
INFO[0000] grpc: addrConn.resetTransport failed to create client transport: connection error: desc = "transport: dial tcp 127.0.0.1:8001: getsockopt: connection refused"; Reconnecting to {"127.0.0.1:8001" <nil>} 
INFO[0001] grpc: addrConn.resetTransport failed to create client tran

glederer007, 2016-11-15:
Hello lancezhangsf,

Currently the loraserver runs only when lora-app-server is running as well. Have you checked the FAQ of loraserver and lora-app-server?  

https://docs.loraserver.io/loraserver/frequently-asked-questions/

https://docs.loraserver.io/lora-app-server/frequently-asked-questions/

Regards,

lancezhangsf, 2016-11-16:
hi  brocaar:
thank you very much for your answer,i can rcv the data from gw, and loraserver show the data,but app-server can't show the data ,the log show some error,    i user eu_433,but some other errors happen ,

11月 16 17:13:41 zhangsf-virtual-machine lora-app-server[13886]: time="2016-11-16T17:13:41+08:00" level=info msg="grpc: addrConn.resetTransport failed to create client transport: connection error: desc = \"transport: dial tcp 127.0.0.1:8000: getsockopt: connection refused\"; Reconnecting to {\"127.0.0.1:8000\" <nil>}"
11月 16 17:13:47 zhangsf-virtual-machine lora-app-server[13886]: time="2016-11-16T17:13:47+08:00" level=info msg="handler/mqtt: publishing data-up payload" topic="application/526973696e674846/node/4786e6ed002f0047/rx"
11月 16 17:13:47 zhangsf-virtual-machine lora-app-server[13886]: time="2016-11-16T17:13:47+08:00" level=info msg="transport: http2Client.notifyError got notified that the client transport was broken EOF."
11月 16 17:13:47 zhangsf-virtual-machine lora-app-server[13886]: time="2016-11-16T17:13:47+08:00" level=info msg="transport: http2Client.notifyError got notified that the client transport was broken read tcp 127.0.0.1:59108->127.0.0.1:8000: read: connection reset by peer."
11月 16 17:13:47 zhangsf-virtual-machine lora-app-server[13886]: time="2016-11-16T17:13:47+08:00" level=info msg="grpc: addrConn.resetTransport failed to create client transport: connection error: desc = \"transport: dial tcp 127.0.0.1:8000: getsockopt: connection refused\"; Reconnecting to {\"127.0.0.1:8000\" <nil>}"

lancezhangsf, 2016-11-17:
Hi，i  have resolve the problem,   in the branch test_eu433 :   
 file :band_eu433.go   
 not find func:   getRX1FrequencyFunc.

i  hope that it can be update.

brocaar, 2016-11-19:
@lancezhangsf I don't see why you would need the `eu433` config as it seems you're on the `CN 470-510` ISM band. Therefore I recommend you to use the pre-compiled binaries. As @glederer007 mentioned, your initial error is caused by the fact that both services need to communicate with each other.

lancezhangsf, 2016-11-21:
hi ,brocaar 
Thank you for your answer.
 433Mhz Just for the project test，470 Mhz is Special frequency of electric meter in china. i need to use 433Mhz for my proj.

brocaar, 2016-11-21:
Ah okay. Please note that the 433 test branch is far behind master as I
wasn't able to test and merge it. If you can confirm these band settings
are correctly defined, I'm more than happy to update and merge it!

aminatamima, 2016-12-14:
Hi Brocar, 
I am a zero who wants to install the network server in my pc for a school project. I have followed all the instructions and all the three components are well set up. But when i launch loraserver before lora-app-server i have the error meaning that loraserver can not be connected to lora-app-server. In return when i start lora-app-server before loraserver i get the same error (meaning lora-app-server can not be connecter to loraserver). They retry to reconnect but i dont know how to detect if they are well reconnected. Do you please have a way to start both of them without having this errors please:

Here is the example of logs meaning it can't be connected to the lora-app-server:

déc. 15 00:30:25 aminata loraserver[11642]: time="2016-12-15T00:30:25+01:00" level=info msg="grpc: addrConn.resetTransport failed to create client transport: connection error: desc = \"transport: dial tcp 127.0.0.1:8001: getsockopt: connection refused\"; Reconnecting to {\"127.0.0.1:8001\" <nil>}"
déc. 15 00:30:31 aminata loraserver[11642]: time="2016-12-15T00:30:31+01:00" level=info msg="grpc: addrConn.resetTransport failed to create client transport: connection error: desc = \"transport: dial tcp 127.0.0.1:8001: getsockopt: connection refused\"; Reconnecting to {\"127.0.0.1:8001\" <nil>}"

brocaar, 2016-12-15:
There is no issue having this error temporarily, as both services know how
to reconnect :)

aminatamima, 2016-12-15:
Ok thank you!
But how do i know if the reconnection went well ?

brocaar, 2016-12-21:
@aminatamima it will log the warning until it re-connected.

closed 2016-12-21 by brocaar