brocaar/chirpstack-network-server #98: ACK not working

mariopraga, 2016-11-16:
Software I use:   ESP-1ch-Gateway-v3.0  + Lora-gateway-bridge + Loraserver + Lora_app_server

The problem:  When sending confirmed messages uplinks or downlinks the gateway freezes. It seems freezing at a **setFreq using:**   step. 

**TTN:**  Below the correct ACK message recived from TTN network 

[8:01:18 PM] Walter Pisani: receivePacket:: LoRa message ready
Packet RSSI: -90 RSSI: -109 SNR: 9 Length: 15 -> 80 27 34 42 ED 0 5 0 1 B 49 79 BC 2F 75 
RXPK:: {"rxpk":[{"tmst":327091317,"chan":0,"rfch":0,"freq":868.099975,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":9,"rssi":-90,"size":15,"data":"gCc0Qu0ABQABC0l5vC91"}]}
sendUdp 1: sent 191 bytes
PKT_PUSH_ACK:: size 4 From 40.114.249.243, port 1700, token: 14A6

{"txpk":{"imme":false,"tmst":328091317,"freq":868.099,"rfch":0,"powe":14,"modu":"LORA","datr":"SF7BW125","codr":"4/5","ipol":true,"size":13,"data":"YCc0Qu0gAgABT+05xw=="}}
data: YCc0Qu0gAgABT+05xw==
txLoraModem::  powe: 14, freq: 868100000, crc: 0, iiq: 0X40
setFreq using: 868100000
start: 327803634, end: 328091317, waited: 287683, delay=0
setFreq using: 868100000
Request::  tmst=328091317 wait=317330
 strict=1 datr=SF7BW125
 freq=868100000 ->868100000
 sf  =7 ->7 modu=LORA powe=14 codr=4/5
 ipol=1
[8:01:28 PM] Walter Pisani: mac tx cnf 1 6868

ok
mac_rx 1

**Lora-gateway-bridge + Loraserver + Lora_app_server**

MQTT : log

 "gateway/5ccf7fffff01f722/tx {"txInfo":{"mac":"5ccf7fffff01f722","immediately":false,"timestamp":1637326882,"frequency":868099975,"power":14,"dataRate":{"modulation":"LORA","spreadFactor":7,"bandwidth":125},"codeRate":"4/5","iPol":null},"phyPayload":"YCc0Qu0gAAC36Xxs"}
"

The gateway freeze. 

Looking forward for receiving any suggestion.

brocaar, 2016-11-19:
Hi @mariopraga 

Given the gateway freezes, it seems there is a bug in the gateway software. If might be caused by a bad transmission request, but then it should be rejected (with some informational logging), not crashing your gateway.

One interesting difference I notice is that the package has been received at a freq of `868.099975` Mhz. TTN truncates this to `868.099` Mhz (thus `868099000` Hz) for the downlink reply and dropping some precision. LoRa Gateway Bridge and LoRa Server are using Hz (to avoid any floating point surprises) thus `868099975` Hz.

For all the channel related mac-commands, the LoRaWAN specs states that the frequency must be a multiple of 100. Maybe that is what your gateway expects? What you could do is transmit some test payloads at different frequencies (e.g. try publishing the above payload, but then with `868099000` as frequency), see if that crashes your gateway too. You can do this by simply publishing the JSON payload to `gateway/5ccf7fffff01f722/tx` (there is no need to setup your node, as it is just to test if your gateway will crash).

Hope that gives you an indication what is going wrong.

mariopraga, 2016-11-22:
Thank you, I will test as soon as possible. 

Just for confirmation, do I need to use mosquitto_pub in order to publish the JSON you suggested? Can you please show me full mosquitto_pub command ? 

thx

mariopraga, 2016-11-23:
I've just finished the test with this payloads:

mosquitto_pub -t "gateway/5ccf7fffff05fbd1/tx" -m "{"txInfo": {"mac": "5ccf7fffff05fbd1","immediately": false,"timestamp": 1637326882,"frequency": 868099975,"power": 14,"dataRate": {"modulation": "LORA","spreadFactor": 7,"bandwidth": 125},"codeRate": "4/5","iPol": null},"phyPayload": "bHVjaWFu"}"

mosquitto_pub -t "gateway/5ccf7fffff05fbd1/tx" -m '{"txInfo": {"mac": "5ccf7fffff05fbd1","immediately": false,"timestamp": 1637326882,"frequency": 868099000,"power": 14,"dataRate": {"modulation": "LORA","spreadFactor": 7,"bandwidth": 125},"codeRate": "4/5","iPol": null},"phyPayload": "bHVjaWFu"}'

mosquitto_pub -t "gateway/5ccf7fffff05fbd1/tx" -m '{"txInfo": {"mac": "5ccf7fffff05fbd1","immediately": false,"timestamp": 1637326882,"frequency": 868099,"power": 14,"dataRate": {"modulation": "LORA","spreadFactor": 7,"bandwidth": 125},"codeRate": "4/5","iPol": null},"phyPayload": "bHVjaWFu"}'

mosquitto_pub -t "gateway/5ccf7fffff05fbd1/tx" -m '{"txInfo": {"mac": "5ccf7fffff05fbd1","immediately": false,"timestamp": 21361234,"frequency": 868099975,"power": 14,"dataRate": {"modulation": "LORA","spreadFactor": 7,"bandwidth": 125},"codeRate": "4/5","iPol": true},"phyPayload": "bHVjaWFu"}'

Same problem, gateway enter in a long loop ( so I confirm is not freeze ). 

  delay(15);          // ms delay() including yield, slightly shorter
  waitTime= tmst - micros();
 }

Some personal debug: 

---startTime: 53574684
---txDelay: 0
---tmst: 120274851
---waitTime: 66700166
---loop waitTime: 66684940
---loop waitTime: 66669840
---loop waitTime: 66654771
---loop waitTime: 3983622026
---loop waitTime: 3983606931
---loop waitTime: 3983591818
---loop waitTime: 3983576727
---loop waitTime: 3983561651
.......
....
...
---loop waitTime: 3979924128
....
.....

I can't understand the logic, maybe  uint32_t waitTime = tmst - micros(); is the problem of long loop ? 

Example: 120274851 - 301499695 = 4113742451 ????

brocaar, 2016-11-25:
The timestamp you're looking at is an unsigned int32. It can be best visualised as a wheel of all possible numbers or a clock.

Given `2^32 = 4294967296`, it means that an `uint32` can contain the values `0 - 4294967295`. Thus `4294967295 + 1 = 0` and `0 - 1 = 4294967295`.

See: https://play.golang.org/p/_7YU3g6P7p

Maybe it is a good idea to open an issue at https://github.com/things4u/ESP-1ch-Gateway-v3.0? It sounds like the gateway software hangs on some data it can't process.

closed 2017-03-15 by brocaar

den3y, 2022-01-08:
I had the exact same problem with the frequency being 868.099975.
I did some research and found out this is related to the ftoa-function in _utils.ino of the ESP-1ch-Gateway.
So in the end for my case the topic was not related to Chirpstack but the 1CH-Gateway.

Finally the pull request from @tfuk did the trick.
I simply updated the _utils.ino and _txrx.ino accordingly and it worked fine.

https://github.com/things4u/ESP-1ch-Gateway/pull/100/commits