brocaar/chirpstack-network-server #100: Make node-session TTL configurable / don't expire for ABP nodes

frederikvannieuwenborg, 2016-11-17:


edjeed, 2016-12-02:
Suggestion:
If no sessiondata for the abp device is available in redis, check the postgres database (only if the 7 MSB of the devaddr are equal to seven LSB of the NetID). 
This way it is also easy to modify the database directly to add/modify/delete devices by a simple script. 

This way it will also be possible to automatic add new devices with a valid combination of keys and id's (NetID, network and application) to the database. All ABP devices should have the same keys, and can be identified by their devaddr, deveui.

brocaar, 2017-05-18:
This will be fixed in 0.17.1 by adding support to configure the node-session TTL.

commit 85e8c5c9 (2017-05-18): Make node-session ttl configurable.

Closes #100.
closed 2017-05-18 by brocaar