brocaar/chirpstack-network-server #101: RN2903 Update

jcwren, 2016-11-27:
I would do the clone and pull request thing, but I have very poor luck any time I do anything git related.

On the page for the RN2903 provisioning, https://docs.loraserver.io/lora-app-server/activating-nodes/#rn2483-rn2903, a couple of extra commands are necessary.

All unused channels should be disabled with 'mac set ch status <channel_id> off', where channel_id is 8..71, assuming you're using channels 0..7. This should be followed by a 'mac save' command to save the channel status to EEPROM.

Secondly, any time a 'mac save' is done, 'mac set devaddr 00000000' should first be issued. Not doing so will result in the 'mac join otaa' command failing. Per the RN2903 manual, "If this parameter was previously saved to user EEPROM by issuing the mac save command, after modifying its value, the mac save command should be called again."

A 'mac save' must be done any time the application key ('mac set appkey'), application EUI ('mac set appeui'), or the device EUI ('mac set deveui') is issued. This is of course also true if other parameters are changed, but so far my experience has shown I've not needed to meddle with the other parameters yet.

Anyone who is this far along probably has figured out that commands to the RN2903 must be terminated with CR + LF, not CR alone. It also helps to turn local echo on.

brocaar, 2016-12-01:
Cheers, I'll add this to the docs soon!

brocaar, 2016-12-05:
I've added your feedback to the docs (https://github.com/brocaar/lora-app-server/commit/0ef345a305985d42355e74f029c6a5a1e4bbd99a). It might take some hours before the docs are updated (because of cache invalidation).

closed 2016-12-05 by brocaar