brocaar/chirpstack-network-server #104: Port settings not working from systemd?

ctowen, 2016-12-02:
Unfortunately, our system already has services running on port 8000 and 8001 so I set out to change the ports that the lora-server and api run on, but it doesn't work when running from systemd.

When I start the loraserver with the following command, everything works as expected and the service starts on the specified port:

/opt/loraserver/bin/loraserver --net-id "010203" --band "EU_863_870" --bind "127.0.0.1:**5560**" --redis-url "redis://localhost:6379" --as-server "127.0.0.1:**5561**" &

However when I start the loraserver as a service as described in your instructions using systemd and systemctl, It tries to find the app-server on port **8001**.  Is it possible there is a bug and it isn't wired correctly in the code to use the port from the configuration file when launching as a service?  Or can you check my settings to make sure they are correct?

I based them off the documentation here:
https://docs.loraserver.io/loraserver/getting-started/

[Service]
Environment="NET_ID=010201"
Environment="BAND=EU_863_870"
Environment="BIND=127.0.0.1:**5560**"
Environment="REDIS_URL=redis://localhost:6379"
Environment="GW_MQTT_SERVER=tcp://localhost:1883"
Environment="AS_SERVER=127.0.0.1:**5561**"

Dec 02 00:27:21 WR-IDP-991C loraserver[3653]: time="2016-12-02T00:27:21Z" level=info msg="grpc: addrConn.resetTransport failed to create client transport: connection error: desc = \"transport: write tcp 127.0.0.1:45528->127.0.0.1:**8001**: write: broken pipe\"; Reconnecting to {\"127.0.0.1:8001\" <nil>}"

brocaar, 2016-12-02:
Please note that the components don't read the config file, the config file is used by systemd to expose the config through environment variables.

You could also test these settings by starting `loraserver` like this:

```bash
NET_ID=010201 BAND=EU_863_870 /opt/loraserver/bin/loraserver
```

If that works with, it means that the config is not correctly loaded (or refreshed) by systemd.

ctowen, 2016-12-02:
Thank you - that makes sense.

I still think there is an issue though.  I don't believe it is getting the port from the env variable.  See below:

root@WR-IDP-991C:~# NET_ID=010201 BAND=EU_863_870 BIND=127.0.0.1:**5560** REDIS_URL=redis://localhost:6379 GW_MQTT_SERVER=tcp://localhost:1883 AS_SERVER=127.0.0.1:**5561** /opt/loraserver/bin/loraserver
INFO[0000] starting LoRa Server                          band=EU_863_870 docs=https://docs.loraserver.io/ net_id=010201 version=0.12.3
INFO[0000] setup redis connection pool                   url=redis://localhost:6379
INFO[0000] backend/gateway: connecting to mqtt broker    server=tcp://localhost:1883
INFO[0000] connecting to application-server              ca-cert= server=127.0.0.1:8001 tls-cert= tls-key=
INFO[0000] no network-controller configured
INFO[0000] starting api server                           bind=127.0.0.1:5560 ca-cert= tls-cert= tls-key=
INFO[0000] backend/gateway: connected to mqtt server
INFO[0000] backend/gateway: subscribing to rx topic      topic=gateway/+/rx
INFO[0000] grpc: addrConn.resetTransport failed to create client transport: connection error: desc = "transport: write tcp 127.0.0.1:33023->127.0.0.1:**8001**: write: broken pipe"; Reconnecting to {"127.0.0.1:8001" <nil>}

brocaar, 2016-12-03:
Ah you're right. I see there is a typo in the env variable config. This has been fixed in the master branch and will be included in the next release. For now, use one of these pre-compiled binaries I made for you: https://www.dropbox.com/sh/nawdaw666ls04y8/AABqCPqNrA4rET8txvLzEcwra?dl=0

closed 2016-12-03 by brocaar

ctowen, 2016-12-04:
Thank you sir - congratulations on this project!  The work you have done here is very impressive.