brocaar/chirpstack-network-server #106: Loraserver is always restarting on incoming message from lopy.

maximilianschwab, 2016-12-11:
Hi guys,

i received my new Lopy (https://www.pycom.io/product/lopy/) a few days ago. Now i tried them out with the loraserver. Whenever i send a message to the server i get following log output:

Dez 11 18:44:26 v22016103605938256 loraserver[27239]: time="2016-12-11T18:44:26+01:00" level=info msg="backend/gateway: rx packet received"
Dez 11 18:44:26 v22016103605938256 loraserver[27239]: time="2016-12-11T18:44:26+01:00" level=info msg="packet(s) collected" dev_eui=70b3d5499aa53363 gw_count=1 gw
_macs=aa555a0000000101 mtype=UnconfirmedDataUp
Dez 11 18:44:26 v22016103605938256 loraserver[27239]: time="2016-12-11T18:44:26+01:00" level=info msg="rx info sent to network-controller" dev_eui=70b3d5499aa5336
3
Dez 11 18:44:26 v22016103605938256 loraserver[27239]: time="2016-12-11T18:44:26+01:00" level=info msg="node-session saved" dev_addr=0752377d
Dez 11 18:44:26 v22016103605938256 loraserver[27239]: panic: runtime error: index out of range
Dez 11 18:44:26 v22016103605938256 loraserver[27239]: goroutine 26 [running]:
Dez 11 18:44:26 v22016103605938256 loraserver[27239]: panic(0x861880, 0xc42000c0d0)
Dez 11 18:44:26 v22016103605938256 loraserver[27239]:         /home/travis/.gimme/versions/go1.7.linux.amd64/src/runtime/panic.go:500 +0x1a1
Dez 11 18:44:26 v22016103605938256 loraserver[27239]: github.com/brocaar/loraserver/internal/downlink.getDataDownTXInfoAndDR(0xc42011f000, 0xb155c0, 0xc42019e9c0,
 0x30201, 0xb176a0, 0xc420022090, 0xb15580, 0xb56c40, 0xaaaadede7d375207, 0x49d5b37034000000, ...)
Dez 11 18:44:26 v22016103605938256 loraserver[27239]:         /home/travis/gopath/src/github.com/brocaar/loraserver/internal/downlink/data.go:57 +0x1fd
Dez 11 18:44:26 v22016103605938256 loraserver[27239]: github.com/brocaar/loraserver/internal/downlink.SendDataDownResponse(0xc42011f000, 0xb155c0, 0xc42019e9c0, 0
x30201, 0xb176a0, 0xc420022090, 0xb15580, 0xb56c40, 0xaaaadede7d375207, 0x49d5b37034000000, ...)
Dez 11 18:44:26 v22016103605938256 loraserver[27239]:         /home/travis/gopath/src/github.com/brocaar/loraserver/internal/downlink/data.go:131 +0x17c
Dez 11 18:44:26 v22016103605938256 systemd[1]: loraserver.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
**Dez 11 18:44:26 v22016103605938256 systemd[1]: loraserver.service: Unit entered failed state.
Dez 11 18:44:26 v22016103605938256 systemd[1]: loraserver.service: Failed with result 'exit-code'.
Dez 11 18:44:26 v22016103605938256 systemd[1]: loraserver.service: Service hold-off time over, scheduling restart.
Dez 11 18:44:26 v22016103605938256 systemd[1]: Stopped loraserver.
Dez 11 18:44:26 v22016103605938256 systemd[1]: Started loraserver.**
Dez 11 18:44:26 v22016103605938256 loraserver[27252]: time="2016-12-11T18:44:26+01:00" level=info msg="starting LoRa Server" band="EU_863_870" docs="https://docs.
loraserver.io/" net_id=010203 version=0.12.3
Dez 11 18:44:26 v22016103605938256 loraserver[27252]: time="2016-12-11T18:44:26+01:00" level=info msg="setup redis connection pool" url="redis://localhost:6379"
Dez 11 18:44:26 v22016103605938256 loraserver[27252]: time="2016-12-11T18:44:26+01:00" level=info msg="backend/gateway: connecting to mqtt broker" server="tcp://l
ocalhost:1883"
Dez 11 18:44:26 v22016103605938256 loraserver[27252]: time="2016-12-11T18:44:26+01:00" level=info msg="connecting to application-server" ca-cert= server="127.0.0.
1:8001" tls-cert= tls-key=
Dez 11 18:44:26 v22016103605938256 loraserver[27252]: time="2016-12-11T18:44:26+01:00" level=info msg="no network-controller configured"
Dez 11 18:44:26 v22016103605938256 loraserver[27252]: time="2016-12-11T18:44:26+01:00" level=info msg="starting api server" bind="127.0.0.1:8000" ca-cert= tls-cer
t= tls-key=
Dez 11 18:44:26 v22016103605938256 loraserver[27252]: time="2016-12-11T18:44:26+01:00" level=info msg="backend/gateway: connected to mqtt server"
Dez 11 18:44:26 v22016103605938256 loraserver[27252]: time="2016-12-11T18:44:26+01:00" level=info msg="backend/gateway: subscribing to rx topic" topic="gateway/+/
rx"

Has someone a good hint form me how to solve the problem?

Regard,
Max

brocaar, 2016-12-11:
That doesn't look any good and might possibly a bug. Could you share your configuration (e.g. which band you're on etc...)?

maximilianschwab, 2016-12-11:
Hi @brocaar,

thank you for the quick response. 
I'm working on the EU_863_870 band and havent set any other custom configuration.
The loraserver version is 0.12.3.

The code i use on the lopy is as following (simply join the network, send a message and listen for incoming data), the join procedure works properly:
```python
from network import LoRa
import socket
import time
import binascii
import sys

# Initialize LoRa in LORAWAN mode.
lora = LoRa(mode=LoRa.LORAWAN)

# create an OTAA authentication parameters
app_eui = binascii.unhexlify('DE DE AA AA 00 00 00 34'.replace(' ',''))
app_key = binascii.unhexlify('06 82 3B E3 D5 29 E0 DC 62 A7 CD 00 D6 4F 8C 2D'.replace(' ',''))

# join a network using OTAA (Over the Air Activation)
lora.join(activation=LoRa.OTAA, auth=(app_eui, app_key), timeout=0)

# wait until the module has joined the network
while not lora.has_joined():
    time.sleep(2.5)
    print('Not yet joined...')

# create a LoRa socket
s = socket.socket(socket.AF_LORA, socket.SOCK_RAW)

# set the LoRaWAN data rate
s.setsockopt(socket.SOL_LORA, socket.SO_DR, 5)

# make the socket non-blocking
s.setblocking(False)

# send some data ("hello")
s.send(bytes([0x48, 0x65, 0x6C, 0x6C, 0x6F]))

# get any data received...
while True:
        data = s.recv(64)
        if data != b'':
            sys.stdout.write("Packet received: ")
            print(data)
```

brocaar, 2016-12-11:
Would you be able to post the output of `mosquitto_sub -v -t "gateway/+/+"`? That should show you all the frames received and transmitted.  Especially the frame received that crashes (+ all the metadata) LoRa Server will be really helpful :-)

brocaar, 2016-12-11:
Actually, what is the value you've configured for this field:
[image]

It should have returned an error message instead of crashing LoRa Server, but I suspect the value you've entered there is not a valid data-rate. Looking at the source, it looks like the "index out of range" happens when doing a lookup on the data-rate.

maximilianschwab, 2016-12-11:
The RX2 data-rate was set to 9, its now working without any restarts.

Thank you!

brocaar, 2016-12-11:
thank you for reporting this issue! the next release (with ADR support) will have this issue fixed :-) See: https://github.com/brocaar/loraserver/commit/31ba1eb6cdd107f86e5ccad7447de5a7ad9cfab1

closed 2016-12-11 by brocaar