brocaar/chirpstack-network-server #112: loraserver restarts 0.13.0

ABalanuta, 2016-12-22:
I have a RPI running the forwarder, loraserver and app-server.
After updating from 0.12.4 to 0.13.0 the loraserver started having the following issue.
I have a heavy flow of packets and this happens right after configuring a node in the app-server.

```
Dec 22 03:22:57 raspberrypi loraserver[5847]: created by github.com/brocaar/loraserver/internal/uplink.HandleRXPackets
Dec 22 03:22:57 raspberrypi systemd[1]: loraserver.service: main process exited, code=exited, status=2/INVALIDARGUMENT
Dec 22 03:22:57 raspberrypi systemd[1]: Unit loraserver.service entered failed state.
Dec 22 03:22:57 raspberrypi loraserver[5847]: /home/travis/gopath/src/github.com/brocaar/loraserver/internal/uplink/uplink.go:59 +0xc0
Dec 22 03:22:58 raspberrypi systemd[1]: loraserver.service holdoff time over, scheduling restart.
Dec 22 03:22:58 raspberrypi systemd[1]: Stopping loraserver...
Dec 22 03:22:58 raspberrypi systemd[1]: Starting loraserver...
Dec 22 03:22:58 raspberrypi systemd[1]: loraserver.service start request repeated too quickly, refusing to start.
Dec 22 03:22:58 raspberrypi systemd[1]: Failed to start loraserver.
Dec 22 03:22:58 raspberrypi systemd[1]: Unit loraserver.service entered failed state.
```

brocaar, 2016-12-22:
Thanks for your report! Could you please share some more `loraserver` logs? You should be able to get them with `journalctl -u loraserver -n 1000` (where `1000` is the number of lines)

Vini74, 2016-12-22:
Dec 22 17:51:16 loraserver systemd[1]: Starting loraserver...
Dec 22 17:51:16 loraserver loraserver[2609]: time="2016-12-22T17:51:16+03:00" level=info msg="starting LoRa Server" band="EU_863_870" docs="https://docs.loraserver.io/" net_id=000000 version=0.13.0
Dec 22 17:51:16 loraserver loraserver[2609]: time="2016-12-22T17:51:16+03:00" level=info msg="setup redis connection pool" url="redis://localhost:6379"
Dec 22 17:51:16 loraserver loraserver[2609]: time="2016-12-22T17:51:16+03:00" level=info msg="backend/gateway: connecting to mqtt broker" server="tcp://localhost:1883"
Dec 22 17:51:16 loraserver loraserver[2609]: time="2016-12-22T17:51:16+03:00" level=info msg="connecting to application-server" ca-cert= server="127.0.0.1:8001" tls-cert= tls-key=
Dec 22 17:51:16 loraserver loraserver[2609]: time="2016-12-22T17:51:16+03:00" level=info msg="no network-controller configured"
Dec 22 17:51:16 loraserver loraserver[2609]: time="2016-12-22T17:51:16+03:00" level=info msg="starting api server" bind="127.0.0.1:8000" ca-cert= tls-cert= tls-key=
Dec 22 17:51:16 loraserver loraserver[2609]: time="2016-12-22T17:51:16+03:00" level=info msg="backend/gateway: connected to mqtt server"
Dec 22 17:51:16 loraserver loraserver[2609]: time="2016-12-22T17:51:16+03:00" level=info msg="backend/gateway: subscribing to rx topic" topic="gateway/+/rx"
Dec 22 17:51:19 loraserver lora-gateway-bridge[2534]: time="2016-12-22T17:51:19+03:00" level=info msg="gateway: received udp packet from gateway" addr=192.168.33.13:46663 protocol_version=1 type=PushData
Dec 22 17:51:19 loraserver lora-gateway-bridge[2534]: time="2016-12-22T17:51:19+03:00" level=info msg="gateway: rxpk packet received" addr=192.168.33.13:46663 data="QCJl2AGAyQACWOw5bBdSBmdcuggJhuAme/YxM5Q=" mac=0000024b080601cb
Dec 22 17:51:19 loraserver lora-gateway-bridge[2534]: time="2016-12-22T17:51:19+03:00" level=info msg="backend: publishing packet" topic="gateway/0000024b080601cb/rx"
Dec 22 17:51:19 loraserver lora-gateway-bridge[2534]: time="2016-12-22T17:51:19+03:00" level=info msg="gateway: sending udp packet to gateway" addr=192.168.33.13:46663 protocol_version=1 type=PushACK
Dec 22 17:51:19 loraserver loraserver[2609]: time="2016-12-22T17:51:19+03:00" level=info msg="backend/gateway: rx packet received"
Dec 22 17:51:19 loraserver loraserver[2609]: time="2016-12-22T17:51:19+03:00" level=info msg="packet(s) collected" dev_eui=333234316d356719 gw_count=1 gw_macs=0000024b080601cb mtype=UnconfirmedDataUp
Dec 22 17:51:19 loraserver loraserver[2609]: time="2016-12-22T17:51:19+03:00" level=info msg="rx info sent to network-controller" dev_eui=333234316d356719
Dec 22 17:51:19 loraserver lora-app-server[2544]: time="2016-12-22T17:51:19+03:00" level=info msg="handler/mqtt: publishing data-up payload" topic="application/0000000000000001/node/333234316d356719/rx"
Dec 22 17:51:19 loraserver systemd[1]: loraserver.service: main process exited, code=exited, status=2/INVALIDARGUMENT
Dec 22 17:51:19 loraserver systemd[1]: Unit loraserver.service entered failed state.
Dec 22 17:51:19 loraserver systemd[1]: loraserver.service failed.
Dec 22 17:51:19 loraserver lora-app-server[2544]: time="2016-12-22T17:51:19+03:00" level=info msg="transport: http2Client.notifyError got notified that the client transport was broken EOF."
Dec 22 17:51:19 loraserver lora-app-server[2544]: time="2016-12-22T17:51:19+03:00" level=info msg="grpc: addrConn.resetTransport failed to create client transport: connection error: desc = \"transport: dial tcp 127.0.0.1:8000: getsockopt: connection refused\"; Reconnecting to {127.0.0.1:8000 <nil>}"
Dec 22 17:51:19 loraserver loraserver[2609]: panic: runtime error: integer divide by zero
Dec 22 17:51:19 loraserver loraserver[2609]: goroutine 33 [running]:
Dec 22 17:51:19 loraserver loraserver[2609]: panic(0x874ac0, 0xc42000c0a0)
Dec 22 17:51:19 loraserver loraserver[2609]: /home/travis/.gimme/versions/go1.7.4.linux.amd64/src/runtime/panic.go:500 +0x1a1
Dec 22 17:51:19 loraserver loraserver[2609]: github.com/brocaar/loraserver/internal/adr.HandleADR(0xc420055380, 0xb3c940, 0xc420154510, 0x0, 0xb3e9e0, 0xc420022628, 0xb3c900, 0xb7e160, 0xc4201d42a0, 0x2, ...)
Dec 22 17:51:19 loraserver loraserver[2609]: /home/travis/gopath/src/github.com/brocaar/loraserver/internal/adr/adr.go:81 +0x13f1
Dec 22 17:51:19 loraserver loraserver[2609]: github.com/brocaar/loraserver/internal/uplink.handleCollectedDataUpPackets(0xc420055380, 0xb3c940, 0xc420154510, 0x0, 0xb3e9e0, 0xc420022628, 0xb3c900, 0xb7e160, 0x2, 0xb39b40, ...)
Dec 22 17:51:19 loraserver loraserver[2609]: /home/travis/gopath/src/github.com/brocaar/loraserver/internal/uplink/data.go:179 +0xbce
Dec 22 17:51:19 loraserver loraserver[2609]: github.com/brocaar/loraserver/internal/uplink.validateAndCollectDataUpRXPacket.func1(0x2, 0xb39b40, 0xc4201787d0, 0x943331f6, 0xc42017a640, 0x1, 0x1, 0x1, 0x0)
Dec 22 17:51:19 loraserver loraserver[2609]: /home/travis/gopath/src/github.com/brocaar/loraserver/internal/uplink/data.go:96 +0x7d
Dec 22 17:51:19 loraserver loraserver[2609]: github.com/brocaar/loraserver/internal/uplink.collectAndCallOnce(0xc420055380, 0xcb0106084b020000, 0xecfeddf67, 0x26561640, 0xb58ee0, 0x4729cec4, 0x33c134e0, 0x1, 0x1, 0x1, ...)
Dec 22 17:51:19 loraserver loraserver[2609]: /home/travis/gopath/src/github.com/brocaar/loraserver/internal/uplink/collect.go:107 +0xf0f
[... cut ...]
Dec 22 17:51:20 loraserver loraserver[2616]: time="2016-12-22T17:51:20+03:00" level=info msg="no network-controller configured"
Dec 22 17:51:20 loraserver loraserver[2616]: time="2016-12-22T17:51:20+03:00" level=info msg="starting api server" bind="127.0.0.1:8000" ca-cert= tls-cert= tls-key=
Dec 22 17:51:20 loraserver loraserver[2616]: time="2016-12-22T17:51:20+03:00" level=info msg="backend/gateway: connected to mqtt server"
Dec 22 17:51:20 loraserver loraserver[2616]: time="2016-12-22T17:51:20+03:00" level=info msg="backend/gateway: subscribing to rx topic" topic="gateway/+/rx"

brocaar, 2016-12-22:
Thanks! I know what the issue is and will release a fix asap (probably tonight).

brocaar, 2016-12-22:
This will be fixed in 0.13.1 which will be available within a few minutes. Thanks again for reporting!

commit 061f8788 (2016-12-22): Fix ADRInterval == 0 check.

Closes #112.
closed 2016-12-22 by brocaar